OnionUI/Onion #587: Changing display menu setting introduces stuttering in games

homopekka2, 2022-10-22:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Exiting a game and then changing a setting in display menu (luminance, hue, saturation or contrast), causes framepacing/stuttering issues when entering the game again. It will not run smoothly anymore. This is very noticable in e.g. Sonic the hedgehog, Sparkster for Sega Genesis (picodrive) or Super Mario Bros for NES (FCEUmm). Restarting the device fixes the issue. 

### Steps to Reproduce

- Run a game 
- Exit game
- Enter display menu and change a setting
- Run game again

### Provide any additional context or information that may help us investigate your issue

This was tested on a new install of Onion OS 4.0.3 using a freshly formatted microSD card.

Aemiii91, 2022-10-22:
Yes this is a known issue of the closed source menu program from miyoo. The fix is restarting the device. The text description in the following prompt will be improved with a note of this for the next version of Onion.

Aemiii91, 2022-10-26:
A note has been added to the english translation

closed 2022-10-26 by Aemiii91