OnionUI/Onion #597: V4.0.3, crash when apply cheat to SFC games

iamleont, 2022-10-30:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

20220418

### Provide a clear and concise description of the issue

Tried to apply cheat to sfc games, all of them crashed and jump back to onion. It only happens to SFC, when I apply cheat to MD, everything works fine.

Also tried to do the same cheat to same sfc game on my 351v, it works fine.

### Steps to Reproduce

Launch sfc game
apply cheat (load and apply)

### Provide any additional context or information that may help us investigate your issue

Nope

Olywa, 2022-10-30:
Onion is not do anything special with regards to cheats, we rely on the functionality of Retroarch and the ([supafaust](https://github.com/libretro/supafaust)) core developers implementation of features. 
The fact cheats are working for you with MD would suggest it's not a system problem but perhaps more a file problem. 

- Are the cheat file and rom file definitely region aligned?
- What format are the SFC roms, if they are zipped perhaps you could try extracting them first? 
   Remember to refresh your roms after extracting (press `SELECT `from the games tab). 

> Also tried to do the same cheat to same sfc game on my 351v, it works fine.
Are the roms and file formats identical on both devices? Is the 351 using the same core as Onion? 
You can check the latter in game by going into RA > Quick Menu > Information.

iamleont, 2022-10-30:
Thanks for quick reply

1. Both cheat and rom file are the same for my Miyoo Mini and RG351V
2. The format of rom is zip, but due to I have lots rom in there, I think I would keep that in zipped file.
3. 351V is using SNES9X as the core for SFC, I think the issue is because of [supafaust](https://github.com/libretro/supafaust), thanks
4. Now I'm trying to replace the core

iamleont, 2022-10-30:
Hi

I turn on the expert mode, and install SNES9X core, now cheat works fine.

Thanks for reminding :), Onion is great!!

Olywa, 2022-10-30:
Glad you have it working with SNES9X. Be aware that Supafaust generally performs a little better on the Mini overall (hence it being the default). 

Supafaust is based on Beetle Snes and looking at the [Retroarch documentation](https://docs.libretro.com/library/beetle_bsnes/) it would appear the cheats are not currently supported.

closed 2022-10-30 by Olywa

iamleont, 2022-10-31:
Yeah, can guess why it is the defaultcore for SNES, I will use the Supafaust for the game which I don't need to turn on cheat :)

You guys did a great job and really help lots of people~ Thanks