OnionUI/Onion #598: Using the M.B.A-mini core to improve the arcade gaming experience on the IREM M92 board

ZOMBIE39, 2022-11-01:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

20220419

### Provide a clear and concise description of the issue

Arcade games on the IREM M92 board cannot have a good gaming experience even on the mame2003plus core.

### Steps to Reproduce

Run arcade games on the IREM M92 board, especially Hook, Ninja Baseball Batman and In The Hunt.

### Provide any additional context or information that may help us investigate your issue

https://github.com/wonder199999/MBA.mini-libretro
The cores that will run on the miyoo mini are available at the link above.
For arcade games on the IREM M92 board, the core is significantly smoother than the mame2003plus with auto-frame skipping enabled, and there are no noticeable side effects to the frame skipping.

Olywa, 2022-11-03:
Hi there,
   Thank for highlighting. Did you try any of the cores from the link yourself?

I was unable to get any of them to run. One of the devs has even tried compiling this core specifically for the Mini and it also would not run. I think the core is designed to work with a GPU (which the Mini does not have). 

Even if we can get this to run I'm not sure it would be included in the release, it's quite a niche request as the majority of the games it supports run OK (either in Mame, FBA2012 or with the CPS1/2/3 and Neo Geo dedicated cores). 

I can of course appreciate the benefit for those few games you mention, to see them run nicer, but for the sake of three games (and only a single request*). I'm not sure it would be hugely beneficial. IF you have been able to make it work I can explain how to set this up as an emulator in your own systems for your use (perhaps you already know?). Out of interest have you checked if those game run (well) with the included FBNEO core, it may be another option? 

*Your request here is sort of a lone voice. It's why we ask at the 'new issue' page that feature requests not be raised as an _issue_. By instead raising it on reddit or discord or even the discussions tab above, you will attract more visibility and a chance to garner support from other users (which can influence dev team decisions on such things if there is more support).

ZOMBIE39, 2022-11-03:
> Hi there, Thank you for highlighting. Did you try any of the cores from the github link yourself?
> 
> I was unable to get any of them to run. One of the devs has even tried compiling this core specifically for the Mini and it also would not run. I think the core is designed to work with a GPU (which the Mini does not have).
> 
> Even if we can get this to run I'm not sure it would be included in the release, it's quite a niche request as the majority of the games it supports run OK (either in Mame, FBA2012 or with the CPS1/2/3 and Neo Geo dedicated cores).
> 
> I can of course appreciate the benefit for those few games you mention, to see them run nicer, but for the sake of three games (and only a single request*), I'm not sure it would be hugely beneficial.
> 
> If you _have_ been able to make it work I can explain how to set this up as an emulator in your own systems for your use (perhaps you already know?). Out of interest have you also checked if those game run (well) with the included FBNEO core, it may be another option?
> 
> *Your request here is sort of a lone voice. It's why we ask at the 'new issue' page that feature requests not be raised as an _issue_. By instead raising it on the reddit sub or the discord channel (the onion team are very active in both) or even in the discussions tab above, you will attract more visibility and a chance to garner support from other users (which can influence dev team decisions on such things if there is sufficient support).

I have uploaded videos of running the game to YouTube.
https://youtu.be/_s8uQ6nKH1k
I'm using this core in the screenshot, and although it's probably not exactly compatible with the miyoo mini's hardware, like maybe not having NEON enabled, I did boot this core and run the game.
I'm guessing that your inability to run it may be due to the wrong version of the rom, are you using FBA version 02.97.38 or a rom greater than or equal to mame version 0.172?
This core has a significant boost for games that the mame2003plus core can't run smoothly, the more obvious ones being the IREM M92 board games, 13 of them in total. Of course, since the MBA.mini core driver itself is streamlined, apart from these 13 games, the chances of having other games supported by the MBA.mini core but not running smoothly on the mame2003plus core are not very good, so the improvement over other arcade cores is really limited.
The FBNEO core is definitely capable of running these games, but not suitable. fbn is much more hardware demanding, and if it's a game that can't run smoothly on the mame2003plus core, the FBNEO core will be even less capable.
I have not used the discord channel, the platform is relatively new to the Chinese, I can try to use it afterwards.
[image]
[image]

ZOMBIE39, 2022-11-03:
Also, the mame2003plus core that comes with the onion system seems to have no effect after setting the frame skipping. I also tried the mame2003plus core based on the latest code and it also has problems with auto-frame skipping on the miyoo mini, the screen gets stuck and doesn't move after enabling it, but these are probably problems with the core itself, which is why I found the MBA.mini core to compensate.

Aemiii91, 2022-11-03:
Thanks for the video, looks like it's running well!

Olywa, 2022-11-03:
Thanks for additional info. Which version of the core from the github releases were you able to make work? I assume the 'linux_generic_arm' release? 

I had tried a couple of the release (and also the specific Mini build by one of our developers - which he also tried unsuccessfully). In each case, when launching directly from RetroArch, after selecting a rom, RetroArch would recommend all other Arcade emulators but not this one.

I can try setting it up as a system so the rom is 'forced' to use the core (maybe this is what you did?).

> The FBNEO core is definitely capable of running these games, but not suitable. fbn is much more hardware demanding, and if it's a game that can't run smoothly on the mame2003plus core, the FBNEO core will be even less capable.
While I 100% agree with you here re FBNEO being ore demanding, we do have some users who have said FBNEO performs very well for many games. Have you tried it on the Mini with these games?

In any case the point stands that this is a niche request so having a discussion with other users (on one of the suggested platforms), to garner support for this to be included, would be in your interest. For now you can add this core to your systems on your own device by copying an existing system folder inside \Emu\ and editing the launch.so and config.json appropriately.

ZOMBIE39, 2022-11-03:
> Thanks for additional info. Which version of the core from the github releases were you able to make work? I assume the May 2019 'linux_generic_arm' release?
> 
https://github.com/wonder199999/MBA.mini-libretro/releases/download/05-15-2019/mba.mini_linux_generic_arm_hardfloat.tar.gz
The version at this link.
> I had tried a couple of the releases (and also the specific Miyoo Mini build by one of our developers - which he also tried unsuccessfully). In each case, when launching directly from RetroArch, after selecting the MBA Mini core, and then a rom, RetroArch would recommend all other Arcade emulators but not this one.
> 
RetroArch needs a .info file to tell it about the core when installing it, such as supported file extensions, whether it supports archives, etc., so that RetroArch will recognize the core when it loads the file. However, the Miyoo Mini's RetroArch does not have the ability to install the core (after all, installing the core is not really meaningful on devices that don't have the official buildbot support of RetroArch), so I tried to install the core on the Windows RetroArch, and then overwrote the configuration file retroarch.cfg on the Miyoo Mini, and I was able to read the information in the .info file.
[image]
[image]
[image]
This is the MBA.mini info file that I modified from the mame2010 core's info file.
[mba_mini_libretro.info.zip](https://github.com/OnionUI/Onion/files/9933189/mba_mini_libretro.info.zip)
Then by comparing the configuration files of RetroArch on Windows and Miyoo Mini, I found that changing the option core_info_cache_enable to false will make RetroArch on Miyoo Mini start reading the .info file, but I am not sure if this has any side effects. It does however seem to be an effective remedy and there is a core arduous_libretro on the Miyoo Mini that is also not identified.
[image]
[image]
> I can try setting it up as a system so the rom is 'forced' to use the core (maybe this is what you did?).
> 
> > The FBNEO core is definitely capable of running these games, but not suitable. fbn is much more hardware demanding, and if it's a game that can't run smoothly on the mame2003plus core, the FBNEO core will be even less capable.
> 
> While I 100% agree with you here re FBNEO being more demanding, we do have some users who have reported FBNEO performing very well for many games. Have you tried it on the Mini with these games?
> 
Yes, I've tested all of these games with the FBNEO core and they all lagged unplayably, not even as well as the mame2003plus core.
> In any case the point stands that this is a niche request so having a discussion with other users (on one of the suggested platforms), to garner support for this to be included, would be in your [...]

schmurtzm, 2022-11-04:
Hi, I just saw you last message @ZOMBIE39 ! I didn't know that it could work with an already compiled version 😅 
Anyway I have successfully compiled it for the Miyoo Mini and created a new .info from ... Mame2010 like you !

[Here the compiled core for DL (from last sources)](https://www.dropbox.com/s/bk4dvaxeb5inhoi/MBA-mini.7z).

And here the makefile section (I'm not the best on this subject so do not hesitate to tell me if there are some error or optimisations to do on it 😉 ) : 
```
# MIYOOMINI
else ifeq ($(platform), miyoomini)
   TARGETLIB := $(TARGET_NAME)_libretro.so
   TARGETOS = linux
   CC = /usr/bin/arm-linux-gnueabihf-gcc
   CXX = /usr/bin/arm-linux-gnueabihf-g++
   LD = /usr/bin/arm-linux-gnueabihf-g++
   AR = /usr/bin/arm-linux-gnueabihf-ar
   RANLIB = /usr/bin/arm-linux-gnueabihf-ranlib
   LIBS += -lstdc++ -lpthread
   CCOMFLAGS += -marm -ffast-math -mfloat-abi=hard
   LDFLAGS += -shared -Wl,--gc-sections  -fPIC
   fpic = -fPIC
   SHARED := -shared -Wl,--version-script=src/osd/retro/link.T
   PTR64 = 0
   ARM_ENABLED = 1
   ALIGNED = 1
```
and then : 
`make platform=miyoomini`

ZOMBIE39, 2022-11-05:
@schmurtzm 
I downloaded it, tested it briefly and compared the off-the-shelf core to the one you compiled and found a slight improvement in smoothness, you did a great job!
So it seems that maybe the original core does not have NEON enabled.
I am able to run IREM M92, if you can't, maybe it's the wrong version of the rom, the core uses FBA 02.97.38 romsets, I downloaded it from a Chinese website, don't know how to get this version of rom abroad for now, can discord share pirated rom?

Olywa, 2022-11-05:
We are not permitted to share roms publicly (here or on discord) but we did locate a 02.97.38 set. The IREM games work fine with the original core but neither Schmurtz or myself could get them to run with his build. Very strange that they work for you, maybe there is something different with the roms. I will try to experiment again over the weekend but perhaps you can confirm the contents of a game to ensure we are aligned. This is the zip for nbbatman...

[image]
 
As I say it runs fine on the original core. Out of interest, are you adding the whole set to your rom folder or only the 13 specific roms?

ZOMBIE39, 2022-11-05:
@Olywa 
[image]
Judging from the crc, it should be the same rom.
[image]
Yes, I only added the IREM M92 game. And for the original core, the new compilation is indeed better.

closed 2023-03-30 by Aemiii91