OnionUI/Onion #600: Automatically save screenshot (or state thumbnail) as rom art image if missing

Olywa, 2022-11-08:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3+

### Miyoo Firmware Version

0419 (N/A)

### Provide a clear and concise description of the issue

Have Onion identify where a game does not have artwork (either on launch or exit of the game) and automatically copy a screenshot if one exists, or the state thumbnail to the relevant `/Imgs` folder and rename appropriately. 

### Steps to Reproduce

N/A

### Provide any additional context or information that may help us investigate your issue

N/A

Olywa, 2022-11-09:
Follow up suggestion by reddit user: [Friendly_Rate_6553](https://www.reddit.com/user/Friendly_Rate_6553/)

> It would be great that we can choose from the game selection (after pressing gameswitcher button) we can choose to define the image as boxart. Having it erase automatically every time we use the gameswitcher should be optional

Aemiii91, 2022-11-09:
One problem with this: the screenshot would also be visible to the guest profile, but wouldn't represent their save state.

Olywa, 2022-11-09:
I don't think that's a problem, the intention is not to have a rom image that necessarily matches your current savestate more just to have a rom image where one does not exist already (i.e. gaps in scraping, patched roms).

nettema, 2022-11-21:
I really like that idea, but I think it can be enhanced even more.
Maybe the use of some naming convention (like a certain suffix or postfix) could help to identify whether the pic is boxart, taken from saveState or screenshot, whether it is updatable or constant, and whether it belongs to the main user or guest. 
In that case, it's possible to configure the behavior of such states, to update the state-art every time on game exit or set it once and keep it forever + distinguish guest arts from main user arts.