OnionUI/Onion #611: Dosbox Pure 0.9.7 slow. OK with version 0.21

bloodykean, 2022-11-14:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1dev-alpha4

### Miyoo Firmware Version

April 2022

### Provide a clear and concise description of the issue

I have tested some games with the new updated core Dosbox Pure version 0.9.7 and some games are from default settings very slow.
I changed many core settings and nothing really helps. The only thing that helps is when you change core performance from auto to max and then it also struggles.
The game i tested is Sid Meier's Colonization and it unplayable with the new version. 
I will test some more games, if there is a standard issue with the updated core. 
Anyone with the same issue?

### Steps to Reproduce

Test Colonization DOS game. Very slow. Unplayable. 

### Provide any additional context or information that may help us investigate your issue

none.

bloodykean, 2022-11-15:
I agree. I have tested several games and come to conclusion that it would be good to keep the old version active and the new version for backup. Some games work probably better with the new core (perhaps Fallout?).

schmurtzm, 2022-11-24:
Some additional information : 
Overflask has confirmed that the new version is a lot slower :

> About DOSbox Pure, it is much slower. On 4.0.3 it was running at 100-97% for Master of Orion 2, Novastorm, and Terminal Velocity. Now it's 50-87%. When I specify the machine type , it's worse.
> Also I had renamed my .exe files with full names to have correct names in Favorites like Master of Orion 2.exe instead of MOO2.EXE. On 4.0.3 it was not a problem, but in 4.1 it gives me an illegal file name error

Dosbox 0.21 seems faster and easier to configure so it will still by default.
We keep Dosbox 0.9.7 for users who want to benefits to the fixes & improvements of this version (e.g. bypass the limitation of 16 keyboard bindings with mouse mode).

So it will be possible to switch the default core to v0.9.7 thank to GLO menu (Game List Options) . This feature will be available in next release.

schmurtzm, 2022-11-28:
Hi again, for beta testers, neon was missing from my dosbox-pure 0.9.7 build, it explains why it was so slow.
New commit here : 
https://github.com/OnionUI/Onion/commit/9c672a5a6aee59fbfde919995e71b3b05db3abe3

Now performance should be more similar to the old 0.21. That's a good news because it's not due to a more accurate core and we can now benefits of the new features without a big impact on performances.