OnionUI/Onion #622: Bugs on news install ?

7zxkv, 2022-12-10:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

20220419

### Provide a clear and concise description of the issue

Hello, 

1/ I have no games displayed in the official core: supergrafx (the games are in the sgfx directory of the roms directory), any idea?

2/ Otherwise I activated the expert mode, the problem is that I see absolutely nothing once in this mode, black screen and not a single category or image, any idea?

thanks

### Steps to Reproduce

new install...

### Provide any additional context or information that may help us investigate your issue

new install...

Aemiii91, 2022-12-10:
Please don't open issues about errors in your setup or support questions. You can use reddit, discord, or GitHub discussions for that.

7zxkv, 2022-12-10:
Except that it is a new installation of 0. without any other modification. is it really me? if you have an idea, you can answer, it may help not only me..

Aemiii91, 2022-12-10:
I would like to help you. I'm just telling you this is not the place. It would help others even more if you asked in the right place.