OnionUI/Onion #641: L and R buttons move when restoring snes game

fringd, 2022-12-16:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

My L and R buttons are getting mapped differently when I restore directly into a game on power on.. this is with the new game switcher which i otherwise love love love.

they move from L1/R1 to L2/R2... not the end of the world, but it screws me up playing hard games.

### Steps to Reproduce

play an snes game.
shut down
power back
notice L and R buttons moved

### Provide any additional context or information that may help us investigate your issue

it's not 100% of the time, and usually if i exit back to the menu and start the game again it fixes it. I've tried to change my button mapping when this happens, but if i do it screws up when i start the game normally.

Is there something different about how the configurations/mappings get loaded when we load directly into a game?

Aemiii91, 2022-12-16:
Did you previously use MainUI's key mapping (in the Settings tab) to remap the triggers?

Is there a correlation between this happening and you using a save state from before updating to V4+?

What happens if you select Restart in the RA quick menu? Are they still swapped?

Did you use the swap setting in Tweaks -> Advanced?

fringd, 2022-12-16:
i've messed with a lot of settings to try to fix this issue, but i think before it showed up i didn't do much.

fringd, 2022-12-16:
swap in the tweaks is off.

fringd, 2022-12-16:
i tried clearing the maps from the retroarch app settings menu

fringd, 2022-12-16:
and clearing them in the snes emulator by hittting the menu button and going to configuration

Aemiii91, 2022-12-16:
That would probably make matters worse since it resets to RA defaults and not Onion defaults - use `Tweaks -> Advanced -> Reset settings` for that

fringd, 2022-12-16:
in the retroarch app
 L is button 10
 R is button 11.
in the snes app using select-menu 
 L button 10 is mapped to L
 R button 11 is mapped to R
 Active Remap file is N/A

Aemiii91, 2022-12-16:
That sounds correct. Can you try resetting MainUI settings (via Tweaks)?

fringd, 2022-12-16:
actually now it's happening when i start without restoring... if i go into the setting with select-menu and update L, R when I hit my L1 and R1 buttons it shows 12/13... which i think are the codes for L2/R2

fringd, 2022-12-16:
yep doing that now...

fringd, 2022-12-16:
okay i think that seems to fix it...

Aemiii91, 2022-12-16:
Great! I'm closing this then 👍

closed 2022-12-16 by Aemiii91