OnionUI/Onion #647: Sound is very told (and distor) where that puss START+R/L

Ninoh-FOX, 2022-12-23:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

adjusting the brightness of the screen with select combo keys i saw that the same thing happened with the sound volume when using start combo keys, now the problem is that I can not adjust it back to the factory volume, whenever I reboot and start a game the volume It is at maximum forcing me to always use the button combination to lower it due to the distortion it causes in the speaker.  I would like to return to the factory volume but I can't find the way.

### Steps to Reproduce

Simplelly change the volumen sound with start/RL combo keys and reboot the console or load a game.

### Provide any additional context or information that may help us investigate your issue

I am lock for a config file for reset this but I don't found nothing.

Aemiii91, 2022-12-23:
The audio boost (`START+R2`) is not saved anywhere. It's always reset when you turn off the device.

You can also reset the audio boost by pressing `START+L2+R2`.

Is the volume wheel turned all the way down? (Just before it's completely silent)

Ninoh-FOX, 2022-12-25:
the volume wheel has never given me problems, it's when I found this function by chance trying to change the brightness, strangely the volume does not reset when I restart, it stays so high that it distorts and I have to lower the volume wheel or lower it with start +L2 until it stays at a suitable decibel.  I'll try pressing start+L2+R2 to see if it corrects itself and I'll comment.

closed 2022-12-25 by Ninoh-FOX