OnionUI/Onion #674: Recent & Favorites Game Info View Not Displaying Screenshot

f8less, 2023-01-27:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1.0 Alpha

### Miyoo Firmware Version

202301041337

### Provide a clear and concise description of the issue

Pressing _Right >_ on the D-pad  when in _Recent_ or _Favorites_ menu, to display the individual _Game Info_ view, only displays the default/fallback image and not the screenshot from Imgs, as it does when viewing _Game Info_ from the _Games_ or _Expert_ menu.

### Steps to Reproduce

Press _Right >_ on the D-pad  when in _Recent_ or _Favorites_ menu, to display the individual _Game Info_ view

### Provide any additional context or information that may help us investigate your issue

Miyoo with Boya chip.

Aemiii91, 2023-01-27:
There's a tool in Apps -> Tweaks -> Tools to fix favorites thumbnails.

closed 2023-01-27 by Aemiii91

dre10g, 2023-07-23:
Recent and Favourites for any game in subfolders are not showing images.
I have successfully setup my arcade section with 2 subfolders like this:
ARCADE
              FBNEO
              MAME
So they show up and boot up fine with their individual cores according to this this wiki..
https://github.com/OnionUI/Onion/wiki/Combining-different-systems-or-cores

The images show up fine in the Games menu, but when I favourite a game from any of those 2 subfolders, it's image does not show in the the favourites list or recents list.