OnionUI/Onion #676: Theme crashing upon quitting RetroArch for long load-time Arcade games

f8less, 2023-01-31:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1.0 Alpha

### Miyoo Firmware Version

202301041337

### Provide a clear and concise description of the issue

Active theme crashes upon quitting RetroArch ("Quit RetroArch" from RetroArch menu) when playing an Arcade title that take longer to load than most (e.g. - Strikers 1945 II - ~10 second load time), resulting in the loading of the default OnionOS (Silky) theme, but is missing text labels (see screenshot).  Active theme can only be restored by shutting down/restarting.

[image]

### Steps to Reproduce

Launch: Arcade > Strikers 1945 II (Wait the ~10 second load time) > Menu + Select > Quit RetroArch

### Provide any additional context or information that may help us investigate your issue

Tested on 2 identical 202301041337 Boya Transparent Black Miyoo Mini, both running 4.1.0 Alpha.  One running the _less_ theme [256GB SanDisk Extreme Card], the other running the _Miyoo blues_ theme [128GB SanDisk Extreme Card].

jackiemoonsoon, 2023-02-01:
I just wanted to confirm that I have been experiencing the same issue when loading various CPS-3 games. Often the gameswitcher won't work entirely like with games like Street Fighter III: 3rd Strike or 2nd Impact.

Xpndable, 2023-02-02:
I was able to confirm these findings from the conversation today in Discord. Best theory explaining what is happening is that the write cache for the SPI Boya chips, which includes the workaround symlink implementation, is being cleared by the device under heavy load, which causes a read error on the system.json file when coming out of the game and back into MainUI.

Xpndable, 2023-02-14:
Another user in the Discord, danbush, had the same issue, but this time when they launch and exit the Activity Tracker application. Have yet to determine a proximate cause for this, as the heavy load theory isn't really supported in this case, the Activity Tracker application is incredibly light.  Though haven't been able to replicate on my own device with the same specifications:
OnionOS v4.0.4
Boya Chip
Firmware 202301041337

f8less, 2023-02-23:
> I just wanted to confirm that I have been experiencing a similar issue when loading various CPS-3 games (that take a long time to load) and using the gamswitcher. Often the gameswitcher won't load when trying to engage it while playing Street Fighter III: 3rd Strike or 2nd Impact.

While I am not having this issue with the GameSwitcher (just a delayed response time), I am having these games, and Street Fighter III: New Generation (Euro) crash the active theme, as described in the OP, as well.

closed 2023-07-04 by Aemiii91