OnionUI/Onion #683: [Feature Request] - Pac-Man : automatically select emulators when roms are present

schmurtzm, 2023-02-03:
The normal way to add roms in Onion is to use Package Manager app to create different rom folders for each systems and then fill them.
But in many case users are putting their roms first and then activate their emulators after. When you add roms from many different systems it can be laborious to select each emulator one by one in Package Manager.

So we could affect a key to automatically check Emulators with roms.
The complexity is to manage the case when many emulators are available for one system :  verified emulators have priority but when it's only in experimental section I don't know how to manage (select all, select none or we add a flag "recommended" ?)

Aemiii91, 2023-02-03:
I think this should only affect verified systems. Also, how are we gonna check for roms? Scanning for files with extension from each `extlist`?

schmurtzm, 2023-02-03:
I've made a script to manage  that [on Tomato](https://github.com/jutleys/Trimui-Smart-Tomato/blob/main/main/Apps/4-EmulatorsCleaner/EmulatorsCleaner.sh) : it count the number of files in roms folders. It could be a source of inspiration but I think that Package Manager is a better place to do it on Onion ;)

schmurtzm, 2023-10-20:
An alternative idea that could help to manage validated and expert emulators : a third toggle state : when there are available roms it just show a new orange state,  it stays the user choice (manual) to enable it.

[image]

closed 2023-12-11 by schmurtzm