OnionUI/Onion #687: Audio for certain mGBA games not working

dreams897, 2023-02-09:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1-alpha6

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Certain audio is not working properly and some sound effects are not being played in mGBA 1.14 core for certain games such as Wario Land 4, regardless of audio output (oss, sdl, audioio)

### Steps to Reproduce

Open mGBA 1.14 core on 4.1-alpha6 build, load game and notice sound issues occurring, but only on certain games that previously worked in mGBA 1.10

### Provide any additional context or information that may help us investigate your issue

Was working previously in mGBA 1.10 core

dreams897, 2023-03-30:
Update: So this issue still persists in 4.1.1 and with the new core update. Some notable games are Warioland 4 and Pokemon Unbound. Certain sound effects are missing and replaced with a very basic sound. This is a CORE issue and not an issue with Retroarch. The last working core version is 0.9.2

Aemiii91, 2023-03-30:
What if you create in-game saves, delete save states, and reset RetroArch and core configs in `Apps -> Tweaks -> Advanced -> Reset settings...`

dreams897, 2023-04-08:
@Aemiii91 that worked, thank you

closed 2023-04-08 by Aemiii91