OnionUI/Onion #696: [v4.0.4] [ports] unable to add Diablo.port to the ports list

dodikk, 2023-02-20:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.4

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

I was following the guide by `@retrobreeze` to play `diablo` on my `Miyoo Mini v1`.
Except I was adding `Diablo.port`, image and binaries to the new directory which is `<SD root>/ROMS/PORTS/...`
After everything was done, the port did not show up on the `ports collection`, so I am unable to even try launchig it. Which seems wrong. 
P.S. other ports like `doom` and `quake` do work ok.

### Steps to Reproduce

0. Pre-requisites: a `miyoo mini v1` device with firmware v202204190828 and OnionOS v4.0.4
1. download the `DevilutionX` miyoo package from https://github.com/brocky/devilutionX/releases
2. put `devilutionx-onion-os.zip/Emu/PORTS/Binaries/Diablo.port` to `<SD root>/ROMS/PORTS/Binaries/Diablo.port/**`
3. create an empty file `<SD root>/ROMS/PORTS/Diablo.port` (or copy it from the DevilutionX archive)
4. copy the doom port icon from the archive to `<SD root>/ROMS/PORTS/Imgs`
5. Put the sdcard back to miyoo, boot the device and `open the ports "device"`

Expected result: diablo port is on the list and the entry is clickable (so the user is able to try launching it)
Actual result: ports collection is the same as for the fresh install of the Onion OS

### Provide any additional context or information that may help us investigate your issue

N/A

sectune, 2023-02-27:
Follow my instructions here, the port was not updated to the new OnionOS folder structure. You’ll find a walkthrough in my link as well as, alternatively, a patched version (no credits to me for the patched version, I just used google):
https://github.com/Brocky/devilutionX/pull/2#issuecomment-1382951716

dodikk, 2023-02-28:
@sectune thanks for the reply. Somehow the port has appeared on the list after installing other games and refreshing game list (dropping the caches) several times.
P.S. Seems like I had figured out those instructions but... for some reason the game did not show up after the first few "clear cache" actions.
P.P.S. By mistake I have placed a few `sega genesis` games to the `SG-1000` dir. Not sure if that's related to diablo port lookup though.

closed 2023-02-28 by dodikk