OnionUI/Onion #698: Play activity databases

Sichroteph, 2023-02-28:


Sichroteph, 2023-02-28:
I leave it here so we don't forget about it.

The new rule to prevent db corruption is to cancel any time registration if the db is empty. That seems to work.
For this a db file had been created that contains a game called "dummy"

1) This db needs to be the new default in Onion (guest + current user)
2) We can hide the game in the playactivity UI if it is called "dummy"

The solution feels like DIY, but it seems to work and I have no other idea. Exiting the mGBA core tend to cause this corruption a lot in my experience.

haydnkilyan, 2023-04-04:
How do you hide / get rid of the 'dummy' entry..... it is in my activity tracker

closed 2023-07-04 by Aemiii91