OnionUI/Onion #709: Red border on screen, battery always 0%

ym2kmobile, 2023-03-14:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1

### Miyoo Firmware Version

202302181113

### Provide a clear and concise description of the issue

Both Onion V4.0.4 & Onion V4.1 is having this issue.

I had just received my miyoo mini+ today. Everything is fine. However when i boot it up with my Onion OS sd card, the first screen i see, there is a red border around my screen. The red border stay even playing a game which is very annoying. Also the battery is always showing 0%.

When i use the default sd card that came with the unit, or boot up without any sd card, everything is fine. I have attached 2 photo for your reference. One is boot without any sd card. The other is with Onion OS.

Even tried to do a clean Onion OS installation on my sd card. The moment it get pass the package installation section, the red box appear.

[image]

[image]

### Steps to Reproduce

Just use a sd card with Onion OS installed in it. Pop it in and the issue will be seen. 

### Provide any additional context or information that may help us investigate your issue

Suspect is incompatibility issue with the latest 202302181113 firmware. Also notice in the setting, with stock sd card, there are additional settings available. As seen in below screenshot, there is setting like "Wifi", "Fixes", "Factory Reset".

With Oinon SD card insert, all these settings when missing..

[image]

Aemiii91, 2023-03-14:
Onion is not compatible with Mini+ yet

closed 2023-03-14 by Aemiii91

eRaz3r, 2023-07-01:
I get the same issue and I don't understand ist. I thought and read that the miyoo mini plus can also run onion OS. Would you please be so kind and explain this to me?

Aemiii91, 2023-07-01:
@eRaz3r you need V4.2

eRaz3r, 2023-07-01:
Thank you. After I look into the releases, I find the hint that the other versions are not compatible at this moment. Thank you.