OnionUI/Onion #710: Issue Title

YanDevDe, 2023-03-14:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.4

### Miyoo Firmware Version

20230104

### Provide a clear and concise description of the issue

I can see the "miyoo linux games" splash screen and then "onion os 4.0.4" splash screen. After this, the screen is just black. But I can still feel the menu is inside, and I can trigger the "Enter search term" UI by entering the keys randomly. So the frontend is running, but the screen remains black.

### Steps to Reproduce

1. I used the community firmware patch (as Onion Installation Website suggested):
https://drive.google.com/drive/folders/1OvmkKEtSMEWv1RY8LPsbDk0ylWzOg_7j

I used "20230104" as firmware.

2. Putting OnionOS 4.0.4 to MicroSD

3. Start Miyoo Mini, OnionOS is installing, and then I can choose to install which Emulator / Apps (I choose the default)

4. After installation, Miyoo Mini shutdowns and turning on - this is where the issue appears.turns

### Provide any additional context or information that may help us investigate your issue

I'm completely new to Miyoo Mini / OnionOS - but I have been following the installation introductions. I might made a mistake here but I've already tried the steps multiple times.

Miyoo Mini were bricked once, but I managed to unbrick it again from the community unbrick guidelines: https://drive.google.com/drive/folders/1-DVPNO78aIJ78QKZz0BAxLWLpxEYpHX3

YanDevDe, 2023-03-14:
additional info:

UI does show up when charging (I see battery animation from OnionOS) and if I shutdown while charging, Is see the UI of "Shutting Down".

Aemiii91, 2023-03-22:
Can you try upgrading to 4.1.1 and see if it solves the issue?

Aemiii91, 2023-03-30:
Closing this until you get back. We can reopen if the issue persists.

closed 2023-03-30 by Aemiii91