OnionUI/Onion #719: ch language missing characters

081944, 2023-03-21:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1.0

### Miyoo Firmware Version

0419

### Provide a clear and concise description of the issue

change language to chinese(ch.lang),text character In home menu is missing.
The reason is that the default font is missing characters.
[image]

### Steps to Reproduce

Settings->Change language->简体中文(ch language)

### Provide any additional context or information that may help us investigate your issue

The default font(Exo-2-Bold-Italic_Universal.ttf) is incomplete, and some Chinese characters are missing.
It can be solved by changing a good font.

Aemiii91, 2023-03-22:
OK, can you give me a "good" font ?

081944, 2023-03-23:
> OK, can you give me a "good" font ?

[Exo-2-Bold-Italic_Universal.zip](https://github.com/OnionUI/Onion/files/11049076/Exo-2-Bold-Italic_Universal.zip)
haha,my english is not that "good".I added some Chinese characters to the font,now it can be displayed normally
[image]

CyberWW, 2023-04-01:
recommend **Noto SansCJK** from Google and Adobe, which has different types of bold, and is free to use

woyin, 2023-04-09:
> recommend **Noto SansCJK** from Google and Adobe, which has different types of bold, and is free to use

Do you know how to use the ttf font in Onion?

closed 2023-07-04 by Aemiii91