OnionUI/Onion #722: Refresh ROM feature bug: list is not getting updated!

dothackjhe, 2023-03-21:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1.0

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Refreshing the ROM list does nothing. The list is stuck from a previously identified/remembered list. 

For example, I renamed a specific GBA file to "XYZ.gba" from "ABC.gba". But after hitting the refresh rom option, I still get the old name for said GBA file, i.e., ABC.gba.

The same is true when adding or removing a ROM. The list simply is not getting updated.

### Steps to Reproduce

Rename, add, or remove a ROM.
Hit the "refresh rom" option.

### Provide any additional context or information that may help us investigate your issue

The "refresh rom" feature is bugged.

Aemiii91, 2023-03-21:
I need more details, we have 3 different "refresh roms" options, which one is bugging out?
1. Press `SELECT` -> `Refresh all roms` in main menu
2. Press `Y` in a game list (GLO) -> `Refresh roms`
3. Select `~Refresh roms` at the bottom of the list (if you have `List Shortcuts` installed)

Aemiii91, 2023-03-22:
If these shortcuts are still giving trouble, you can try reinstalling "List Shortcuts" via Package Manager

closed 2023-03-30 by Aemiii91