OnionUI/Onion #723: NeoGeo CD

7zxkv, 2023-03-22:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1.1

### Miyoo Firmware Version

20220419

### Provide a clear and concise description of the issue

Hi,

I started some games to test. I wanted to launch a neogeo cd game, impossible to load (it stays on loading about 2min before coming back in the shell, I even thought it had crashed. Clearly the emulation does not launch I would say).).
test twice, no problem with 4.0.4 on the same game I insist (Samsho RPG ).

if I have to I'll upload you exactly the version of the game I'm using and that worked very well until v4.0.4. (just tell me where to do it because obviously it's not supposed to be publicly distributed)

best regards

### Steps to Reproduce

.

### Provide any additional context or information that may help us investigate your issue

.

7zxkv, 2023-03-22:
For your information it was a bin/cue that worked very well. I just turned it into a chd, and it works. It's a good thing that it's in chd but in principle it's still a stopgap, the original problem remains.

Aemiii91, 2023-03-30:
This core hasn't been changed since V4.0, making it almost impossible to know what causes the issue. The next course of action  would've been resetting RA and core configs in Tweaks. If that does nothing, then remove saves.

closed 2023-03-30 by Aemiii91