OnionUI/Onion #734: (MMP) manage RTC sync thanks to NTP when wifi is ON

schmurtzm, 2023-03-28:
A Tweaks app option should allow to set NTP sync time ON/OFF.

We will have to add a field in Tweaks app to set the timezone.

Also we can take a look to the ClockAdjuster and ClockDaemon made by Eggs from the Smart :
https://www.dropbox.com/sh/q8ejb10k5k7xup7/AADy-ZefxidGDu--x33g9NLEa
(Eggs set the timezone in a text file, it could help for some inspiration)

At the same time we could include the the clock app in Tweaks app : 
- Manual set of clock (the equivalent of the "old" Clock app) could be in the "System" section of Tweaks.
- And NTP sync toggle and the timezone could be in the new "Network" section.

So may be 3 tasks to do in Tweaks app for this issue : 

- [ ] Create a NTP toggle 
- [ ] Create a timezone field 
- [ ] May be integrate the actual Time Clock App directly in Tweaks

schmurtzm, 2023-03-28:
Interesting remark from Eggs :  

> The time zone setting could conceivably be automated using sites like these, but I don't know if it is necessary to go that far. (I just did a little searching and found them, so there are probably others out there, and I haven't read the terms and conditions carefully)
> http://ip-api.com/csv/?fields=timezone,offset
> https://ipapi.co/utc_offset/ 
> Perhaps, in bordering towns, etc., the identification can be mistaken.

So if no time zone has been set, we could propose a default one to the user.

Calchan, 2023-03-30:
> Interesting remark from Eggs :
>
>
> So if no time zone has been set, we could propose a default one to the user.

I think using `http://ip-api.com/line?fields=timezone` would be better than an offset since you'd get a real timezone which tracks daylight savings, so you wouldn't have to change the offset every 6 months. That assumes tzdata is available though.

Also, this kind of detection doesn't use GPS but depends on where/how your ISP serves you data from. So localization is indeed approximate but most likely in the same administrative region, thus same timezone too. Errors are still possible but very unlikely. Except behind a VPN or proxy but these users will surely be aware of the potential issue.

anzz1, 2023-04-15:
I can confirm that both options properly track DST for me.
http://ip-api.com/csv/?fields=offset ✅
http://ip-api.com/line/?fields=offset ✅
http://ip-api.com/csv/?fields=timezone,offset ✅
http://ip-api.com/line/?fields=timezone,offset ✅
https://ipapi.co/utc_offset/ ✅

The `ip-api` option would use marginally less battery since SSL negotiation isn't required.

Using HTTPS has also the caveats of 
a) need to keep the certificates updated or skip the certificate check 
b) need to keep openssl updated or future changes to the key-exchange algos will break the ability to make connection. this has happened recently with the move towards `ECDH` from regular `DH` where older software cannot establish SSL connections to sites which don't provide regular `DH` anymore.

While HTTP will simply always work.

Aemiii91, 2023-07-04:
Closing as done.

Related PRs:
- #979 
- #977 
- #964 
- #853

closed 2023-07-04 by Aemiii91