OnionUI/Onion #736: Active theme crashes after exit of GameSwitcher

f8less, 2023-03-31:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1.2

### Miyoo Firmware Version

202301041337

### Provide a clear and concise description of the issue

After performing more elaborate series of actions within the GameSwitcher, upon exit, the active theme crashes, reverting to the default Onion theme, but lacking all text.
[image]

### Steps to Reproduce

To my best of abilities to reproduce it, as far as I can tell, if I launch the GameSwitcher, delete an entry, press right (to skip the next), delete that game, press right twice (to skip the next two games), delete that game, then press B to exit...my theme crashes and exits to the default Onion theme lacking all text.   While the above is seemingly not reproducible, it seems to indicate that more elaborate series of actions, before exiting GameSwitcher, can cause the crash of the active theme.

### Provide any additional context or information that may help us investigate your issue

I tested this deleting 3 entries in succession, and it doesn't seem to cause the crash.  It seems to have to be a more elaborate set of actions, before exiting the GameSwitcher. I have been able to cause this to happen at least 3 times in one evening.

f8less, 2023-03-31:
Same theme crash after just casually playing MegaDrive, Lightening Force - Quest for the Darkstar for a bit, then **Menu** > **B** > **_*theme crash*_**

Aemiii91, 2023-07-04:
Possibly related to the *boya incident*.