OnionUI/Onion #743: Playstation, Ports games not updating afterupate

Tqe-sylverne, 2023-04-05:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1.3

### Miyoo Firmware Version

2022204190828

### Provide a clear and concise description of the issue

I have several playstation games in the PS folder that were present prior to my update to 4.1.3 from 4.0.4.  I can see the games in retoarch and launch them from retroarch but not seen by onionos.    No matter how many times i refresh the folder or do a whole refresh form the root device folder i  the games will not show  in the lineup.

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue

try to copy a group of .BIN extension games into the PS folder and see if onionOS sees them after a rom refresh

### Provide any additional context or information that may help us investigate your issue

these games were present and  functioning prior to the upgrade.  I followed the update instructions as written and no issues during the install.

Aemiii91, 2023-04-05:
Read the release notes.

closed 2023-04-05 by Aemiii91

Tqe-sylverne, 2023-04-05:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
> EXTERNAL EMAIL: This email originated from outside of the organization. Do not click any links or open any attachments unless you trust the sender and know the content is safe.

Found it