OnionUI/Onion #752: Brightness hotkey changes brightness and color palette in GB

Qeynin, 2023-04-09:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2 beta

### Miyoo Firmware Version

20230326

### Provide a clear and concise description of the issue

Select + R2/L2 changes the brightness as intended. R2/L2 also swaps the color palette as intended while playing GB roms. However, when using the brightness hotkeys while playing a GB rom, it changes the brightness and swaps the color palette at the same time. Would it be possible to disable the GB palette swap when the select hotkey is held?

### Steps to Reproduce

Open a GB rom. Hold select + R2 or L2 to change the brightness. It will change the brightness and swap the color palette. 

### Provide any additional context or information that may help us investigate your issue

N/A

Aemiii91, 2023-04-09:
Alternatively, use the new shortcut: <kbd>MENU</kbd>+<kbd>VOL UP</kbd>/<kbd>VOL DOWN</kbd>.

Qeynin, 2023-04-09:
That does indeed work. However you then run into the slightly smaller issue of opening the roms refresh tab or exiting out a game if you don't hit the vol. buttons quick enough.

closed 2023-05-29 by Aemiii91