OnionUI/Onion #754: Shoulder buttons keep swapping

MrWidge, 2023-04-09:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2 beta 

### Miyoo Firmware Version

Latest 

### Provide a clear and concise description of the issue

When only ony PS1 - using the function button to go back to the main menu or using it to load up the most recent game. It keeps switching switching the shoulder buttons. I.e. R1 will become R2 or back and forth. The only way to reset it is to load the game up form the amin menu. 

### Steps to Reproduce

Use the function button to go back to the main menu and then back to load the game. 

### Provide any additional context or information that may help us investigate your issue

N/a

Aemiii91, 2023-04-09:
Try resetting all settings via `Apps -> Tweaks -> Advanced -> Reset settings`

MrWidge, 2023-04-09:
> Try resetting all settings via `Apps -> Tweaks -> Advanced -> Reset settings`

Gave it a try. No success yet. It's only a small issue but it's odd. Sometimes the shortcut commands (fast forward) also swap if it's dependant on a shoulder button.