OnionUI/Onion #761: Hardcore mode should be off by default

withfoam, 2023-04-10:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0-beta-dev

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

When enabling Retro Achievements, the default setting for Hardcore Mode is ON. This breaks auto-resume for games launching from the app switcher. They just reboot each time.

### Steps to Reproduce

- Enable Retro Achievements
- Launch any game for any system
- Press Menu button to go to the app switcher
- Press A button to return to the game
- The game reboots instead of resuming where it was

### Provide any additional context or information that may help us investigate your issue

I enabled Retro Achievements during setup/installation of OnionOS 4.2.0-beta-dev on a new SD card.

Aemiii91, 2023-04-10:
This was already fixed in the latest release

closed 2023-04-10 by Aemiii91

withfoam, 2023-04-10:
Oh, great! Apologies for the churn.