OnionUI/Onion #764: Clicking "Start Net Play Host" Freezes Miyoo Mini Plus

Johnnylokz, 2023-04-11:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.2.0-beta-dev-4c7e2db

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

While using the Miyoo Mini Plus me and my friend decided to play streets of rage 2 on the sega genesis emulator which is powered by picodrive. So we both turned on our Miyoo Mini Plus devices which are running the same exact firmware and same exact Onion version with the same exact rom. Once the game was running one of us was required to be the host while the other player would join in. So I navigated to the Netplay menu and clicked on "Start Netplay Host", immediately after clicking this option I was booted back into the game but everything was frozen. No matter which button I pressed the device remained frozen. The only way I could unfreeze the screen was to hold the power button and shut down the device then turn back on. We tried to initiate Netplay multiple times but it ended with the same results every time. At this point Netplay is unusable, I wanted to bring this issue to your attention and in the future I would like to see the Netplay feature working. I would also like to point out that before installing Onion the Netplay feature was working without any issues.

### Steps to Reproduce

Power on the Miyoo Mini Plus and connect to a wifi network. Afterwards go to the Game section and select the Sega genesis emulator (powered by Picodrive) and select "Streets of Rage 2". Before selecting "1 Player" or "2 Players" navigate to the RetroArch quick menu by holding the select and menu button. Once your are at the quick menu click "B" to get to the main menu. Scroll down and select Netplay, then select Host & finally select "Start Netplay Host"

Afterwards you will be brought back into the game but the device will be frozen

### Provide any additional context or information that may help us investigate your issue

I have tried this with other roms & few different emulators with the same results

karahod, 2023-04-11:
Thanks god me and my brother are not the only ones who has this. Really hoping team can fix this.

Johnnylokz, 2023-04-12:
I hope so tooo, I tested this issue on a few different miyoos just to make sure it was not an isolated problem. Im sure the onion team can get this worked out its just a matter of time now

Olywa, 2023-05-05:
This appears to be a bug in RA under certain configuration scenarios (hence it working in stock but not Onion). 
After identifying the problem, Schmurtz has logged an issue with the RA team, a workaround is also there:
https://github.com/libretro/RetroArch/issues/15248

Additionally he has shared the info on reddit here in response to a post:
https://www.reddit.com/r/MiyooMini/comments/1373ayo/comment/jis00rf/?utm_source=share&utm_medium=web2x&context=3

There will be a solution in a future release.