OnionUI/Onion #766: Create an app to quickly select combinations of overlay + filter + core options

schmurtzm, 2023-04-12:
The idea is to have an app to choose with visual examples some Retroarch presets for emulators. 
Typically you choose GB system , then you have a gallery of proposed presets with screenshots. 

Because a good RA configuration is boring to do for novices and it's hard to have the good combination of settings.

[Jeltron guide](https://github.com/OnionUI/Onion/discussions/708 ) will be very useful to create the first templates.

My idea currently is to start simple : a screenshot and the extraction of the required configuration lines in the core override.
Then when you select a screenshot it will override the "core override" to get the template applied in one second.

For example from a proposed screenshot about GB, you could apply an overlay + a filter + some blending from core option in one click.

rogerioojr, 2023-04-12:
if possible, an option together that manages to change the dimensions of the screen, so that the overlay does not cut the screen as it happens now

schmurtzm, 2023-04-12:
> if possible, an option together that manages to change the dimensions of the screen, so that the overlay does not cut the screen as it happens now

The easiest way is to have dedicated overlays like the one that Jeltron has done for Onion.
However I think that it is possible to set the scale and the offset of the overlay in RA settings so we could imagine a preset which include these settings too.

sinedied, 2023-07-26:
I was going to suggest something like this then I saw this issue :)
What about a simple UI screen where you can choose globally between 3 presets?
1. sharp pixels / no overlays / full screen with aspect ratio
2. screen emulation: grid / scanlines depending of device's screen using overlays or filters (best depending of system) / full screen with aspect ratio
3. real: grid / scanlines + border overlays / integer scaling

Choosing one of these presets would apply settings globally, rather than having to select it for each device. 
The first preset is basically the current default. The preset 2. is what I've currently set up on my device (took me quite some time to find the best combination!), so I could contribute it.

retroNUC, 2023-08-12:
As a potential first step, I would suggest taking Jeltron's guide, making sure it's up to date with the overlay renames from 4.2, and added to the Wiki as a dedicated page.

JakiShen, 2024-01-14:
Add Gamelist.xml file generator please