OnionUI/Onion #767: Overlays often create strange effects in the GS

schmurtzm, 2023-04-12:
[image]

There is an alternative way to make screenshots that I had proposed in the past which is to use the screenshot from RA save state.
The overlay is not displayed but the screenshot is perfect.

However the resolution depend of the core.

May be the rotozoom lib was here for that ?

(Vsync enabling doesn't change this behavior.)

gricardo93, 2023-05-30:
Yes. All my GS previews are like this image or worse. It would be a great improvement if it can be fixed. I also noticed that the savestate preview are perfect.

saskwatch, 2023-06-16:
This is really prevalent with the gamebatte core.

Aemiii91, 2023-07-18:
Fixed with https://github.com/OnionUI/Onion/pull/1011/commits/35bf2426953deaf612a10267644a73d81e08a1d2

closed 2023-07-18 by Aemiii91