OnionUI/Onion #768: implement an "on the fly overclock feature"

schmurtzm, 2023-04-12:
« My game is a little slow , hop a little boost »

The idea is to trigger a quick overclock UI in game. 
The UI could be a simple numeric value modified with a shortcut combination.
Could allows underclock too.

Setting is saved per game.

schmurtzm, 2023-06-02:
Eggs  has implemented this feature in his version of retroarch.
There is also a command line binary to change the speed easily.

wzqj, 2023-06-26:
> There is also a command line binary to change the speed easily.

Can you provide a link to the binary so we can test, at our own risk of course.

schmurtzm, 2023-07-30:
[Here the link](https://www.dropbox.com/sh/hqcsr1h1d7f8nr3/AABtSOygIX_e4mio3rkLetWTa?preview=overclock_test.zip). This command line tool has been made by Eggs.

schmurtzm, 2023-08-01:
We now have it in Onion 4.2.0 , but it's kinda hidden. It works with cpuclock.txt. 
Retroarch will looks in the core config folder first, then the RA root folder.
We still have to implement an interface to make it possible to manage it easily.

Overclock is interesting to improve performance (I have measured something near 20% on some emulators) but underclock is not really interesting for power consumption (see below).  Core voltage have more effect on power consumption. Core voltage is different between MM and MMP, it explains why we can overclock more the MMP.

**Some additional information on it :**  
([Credits Eggs messages on discord](https://discord.com/channels/529983248114122762/1087760354278387722/1122335171383918613) _all this conversation about power consumption is very instructive_)

The content of cpuclock.txt is simply a 4-digit number(MHz), accepted from 100 to 2400
miyoomini becomes unstable beyond about 1700, mini+ becomes unstable beyond about 2000, but check for yourself as there may be individual differences
In case of miyoomini, if the device hangs beyond acceptable limits, the power button will not work and the battery should be removed
For miyoomini+, press and hold the power button more than 6 seconds to shutdown
If set below 1200, the emulator runs at a lower clock than stock (underclock).

**About Power consumption when overclocking:** 

After disconnecting the battery and checking with USB voltage checker (although not sure if this measurement method is correct), I found that the voltage change has some effect on power consumption, but the clock change does not have much effect
og mini.

>  600MHz  900mV 0.171A 0.839W
> 1000MHz  900mV 0.172A 0.844W
> 1200MHz 1000mV 0.186A 0.912W
> 1600MHz 1000mV 0.188A 0.921W

plus (wifi off)

>  600MHz  880mV 0.206A 1.008W
>  600MHz 1100mV 0.239A 1.165W
> 1200MHz  880mV 0.206A 1.008W
> 1200MHz 1000mV 0.225A 1.098W
> 1200MHz 1100mV 0.239A 1.166W
> 2000MHz 1100mV 0.246A 1.199W

**Core voltage**

Core voltage can also be changed with [this tool](https://www.dropbox.com/s/n2d7vqep8cg83ty/axp_test3.zip). It may be interesting to see how much difference it makes at 880mV, which is closest to the recommended minimum on ssd202d specsheet  

> 1100mV `axp 22 19` (mmp default, Absolute Maximum value on specsheet: 1.1V)
> 1000mV `axp 22 14` (og mini is working with this value if I remember correctly)
>  900mV `axp 22 0f` (Recommended typical value on specsheet: 900mV)
>  880mV `axp 22 0e` (Recommended minimum value on specsheet: 870mV)

However, since these are values on the terminal screen, they may be affected when the emulator is actually running.

Fun fact : it is the built-in speaker that affects power consumption the most, and in fact, the headphones are the most effective for making the battery last longer

-----------------------

We can also mention this document from tGecko about Miyoo Mini Plus power consumption : https://www.docdroid.net/0zoYpfO/miyoominipluspower-pdf  or [this one](https://docs.google.com/spreadsheets/d/1dNV36xopDolQnARcC1dXGVXLpM2khyQoo8rTqLupME8/edit#gid=0) which reveals the impact of wifi and brightness.

-----------------------

Interesting commands : 
```
/mnt/SDCARD # cat /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
/mnt/SDCARD # cat /sys/devices/system/voltage/core/voltage_current
```

iPruch, 2023-08-18:
@schmurtzm Interesting. Thanks for the info!

I was giving it a try, but the only way I made it work is putting the `cpuclock.txt` in the RetroArch root folder. I would like to apply this only for mGBA core (at least for now), but don't know how. I tried putting it into `<MySDCard>/Saves/CurrentProfile/config/mGBA` (which is where the cfg and opt files from RA are saved to and read from) with no effect. Any clue?

nunits, 2023-09-10:
It would be wonderful if the clocks and voltages could be adjusted from the onion interface.
Mhz increase and undervolt could be tested, giving performance with excellent battery life.

JakiShen, 2024-01-14:
Add Gamelist.xml file generator please