OnionUI/Onion #769: Make a SDL battery icon instead of a red rectangle for low battery

schmurtzm, 2023-04-12:
The idea is to have something similar a low battery more understandable and not a red rectangle.
MiniUI use a real battery icon in SDL overlay.

schmurtzm, 2023-05-08:
This is the culprit. Displaying the icon here would solve it (although it would flicker). The trouble is that mini has triplebufs. https://github.com/OnionUI/Onion/blob/main/src/common/system/display.h#L142

Credits Eggs ([on discord](https://discord.com/channels/529983248114122762/926552349609164872/1104331119735873586
))

schmurtzm, 2023-05-08:
Just a battery icon drawn in the same way as the red rectangle one here : https://github.com/OnionUI/Onion/commit/efb4821cb2832e32c52e46d329b11c8a89f0c8fc
The function allows to choose the color and the position of the icon.
A real modified SDL library could be even better to avoid flickering but it still a good improvement to the original red frame.

**TO DO :** 
- Do not display it when we are in MainUI (we already have a battery level displayed theorically)
- Send the real battery level in parameter

JakiShen, 2024-01-14:
Add Gamelist.xml file generator please

schmurtzm, 2024-03-14:
Do not display it when we are in MainUI  -> done since 4.3
4.3 also allows to hide the low battery icon from Tweaks.

Note that some user would like to be able to choose the red rectangle instead of the icon (few people but in my opinion but who know how to make themselves heard ;) ), I suppose it was mostly due to the icon interaction with MainUI so it's better now however it could be an option.