OnionUI/Onion #770: Sound for retroarchivements

schmurtzm, 2023-04-12:
This RA feature is nice to have.
It requires to enable audio mixer flag in RA. Impact on performances ? Eggs has an opinion on that.

SoggyElbow, 2023-10-24:
Please add this!!
I understand that an icon/badge might be difficult to achieve but a nice little sound notification would make a huge difference.

SentaiBrad, 2025-03-26:
Hopefully it's not rude to ask, but is there an ETA of restoring Appearance Settings for Achievements? Missing Achievement update notifications (Connection, Found RA's for specific game, Achievement progress updates, etc) does make RA's hunting harder to do at times. It's also helpful getting text updates that a game you just loaded its achievement set properly. I appreciate all your hard work on this project.

Denellyne, 2026-01-18:
About this issue, I saw that the problem is enabling the audio mixer, because it will harshly affect performance, because of that I tried to make it so it would play the unlock sfx when you unlock an achievement but when it does it actually mutes the audio stream before and unmutes after.
Given the sfx is 1 second long that seems reasonable to me, but then again, I couldn't figure it out as I am not familiar with that codebase