OnionUI/Onion #774: Aspect ratio shortcut : behavior improvement

schmurtzm, 2023-04-12:
Thanks to Eggs we have a shortcut to switch aspect ratio quickly : 
[video_driver_apply_state_changes](https://github.com/OnionUI/RetroArch/blob/068241784fcf0ff3da4b33b8fe92bf5736040d74/retroarch.c#L3685)

Requested improvements : 

- The change of the ratio require a manual core override to be saved. May be an automatic save could be nice.
- A notification (RETRO_MESSAGE_TYPE_STATUS) could be interesting to know which is the current ratio setting (the 4 possibilities are exposed in this [official Retroarch feature request](https://github.com/libretro/RetroArch/issues/14495))

0neblock, 2023-06-03:
@schmurtzm I have a PR in retroarch here that does point #2: https://github.com/OnionUI/RetroArch/pull/7

schmurtzm, 2023-06-03:
Nice ! Thanks, I'll take a look !

Aemiii91, 2023-07-04:
Done in:
- #968

closed 2023-07-04 by Aemiii91