OnionUI/Onion #780: WiFi password does not support special characters

AceXintense, 2023-04-15:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0-beta

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

The password input for WiFi setup does not currently allow the use of certain characters that are completely valid in WiFi passwords.

I have a * in my passwords which does not have a supporting key to use on the interface.

### Steps to Reproduce

1. Setup secure WiFi access point with random password generated with special characters
2. Attempt to connect to said SSID
3. Attempt putting password in with *,^¥$€ etc

### Provide any additional context or information that may help us investigate your issue

Please make the input keyboard support all valid WPA2 characters

Aemiii91, 2023-04-15:
This will either get fixed by Miyoo (if they release a fixed MainUI) or if we implement our own wifi menu.

gustavokei, 2023-04-20:
Seems like special characters on the SSID (network name) are not supported as well

iKristoffer, 2023-04-22:
Is there a way to manually add the password in some way as a temporary workaround? 🤔

FDrebin, 2023-04-25:
Ran into the same issue when the password had '$' in it. Ended up making a temp network until it's resolved.

AceXintense, 2023-04-30:
> Is there a way to manually add the password in some way as a temporary workaround? 🤔

https://github.com/OnionUI/Onion/issues/781#issuecomment-1511717316

Was mentioned here that the WPA supplicant could be edited maybe try this? I ended up deploying a new WiFi network for the meantime.

JackThird, 2023-05-03:
same here, I cannot find "=" symbol

zaepfchenmaN, 2023-05-11:
same roblem no " " " on the keyboard,

VadimBrodsky, 2023-05-17:
I was able to connect to my wifi network (has an \* in the password) by editing the configuration.

1. Go to "Apps" -> "Package manager"
2. Install the file explorer app "DinguxCommander"
3. Open "File Explorer"
4. Go to the first line `..` press A
5. Go to the first line `..` press A
6. Go to the line with `appconfigs` press A
7. Go to the line with `wpa_supplicant.conf` press A
8. Edit the network configuration, should look something like this:

```
network={
	ssid="ENTER_YOUR_WIFI_SSID_HERE"
	psk="ENTER_YOUR_WIFI_PASSWORD_HERE"
}
```

9. Center button -> quit  
10. Reboot the device from the settings menu

closed 2023-05-29 by Aemiii91

jsoberg, 2023-06-12:
> I was able to connect to my wifi network (has an * in the password) by editing the configuration.
>
>     1. Go to "Apps" -> "Package manager"
>
>     2. Install the file explorer app [...]

Clarifying note for anyone using this fix, you need to keep the original `ctrl_interface=/var/run/wpa_supplicant` first line for this to work. So, if your SSID is `mywifi` and your password is `password$123`, your `/appconfigs/wpa_supplicant.conf` file should look like this:
```
ctrl_interface=/var/run/wpa_supplicant
network={
	ssid="mywifi"
	psk="password$123"
}
```

ZeFifi, 2023-06-14:
Thanks for the tips but it doesn't work here. I found the file, but since I don't know if it's possible to return to the next line, I wrote everything on a single line: 

`network={ ssid="MY_WIFI_SSID" psk="MY_WIFI_PASSWORD }`

Then I try the OTA update but I get error messages. Could someone help? Thanks a lot!

jsoberg, 2023-06-18:
@ZeFifi did you include the `ctrl_interface=/var/run/wpa_supplicant` line at the top, and then restart the device afterward? I don't think the indentations are required but you should add them just in case (4 spaces), so that your file ends up looking like this:
```
ctrl_interface=/var/run/wpa_supplicant
network={
    ssid="MY_WIFI_SSID"
    psk="MY_WIFI_PASSWORD"
}
```

ZeFifi, 2023-06-21:
Yes, that's what I did, but it didn't work. At the end I changed my WPA password without special characters and it works now.

ZeFifi, 2023-06-22:
Do the ssid and psk values have to be strings because that's what I did? I also added the spaces and I still get the "Fail to connect" message.

mkasberg, 2023-09-10:
Editing `wpa_supplicant.conf` from DinguxCommander worked for me! (With a `.` in the wifi password!)

I removed the `update_config=1` line and removed the `disabled=1` line that my file happened to have.

TEPhillips, 2023-09-12:
I followed these instructions to the letter and now my wifi just endlessly scans :( My password has a ' in it.

ArfyFR, 2025-01-17:
I also faced the problem:
 - I have 2 Wifi at home with same password (with a `&` in the password)
 - with a underscore `_` in the Wifi name (TP_Link), no way to connect as if I was editing `wpa_supplicant.conf`
 - the other ok by editing `wpa_supplicant.conf`