OnionUI/Onion #781: Cannot connect to hidden SSIDs

AceXintense, 2023-04-15:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0-beta

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

Currently there is no support to connect to hidden SSIDs this can cause issue when attempting to connect to WiFi.

* Context: Hidden SSIDs do not increase security but does allow it to be hidden from everyday people. Hiding an SSID cannot and should not be used as a security mechanism.

### Steps to Reproduce

1. Create SSID that is hidden.
2. Attempt to connect to it via Onion OS interface.
3. No option to manually add it.

### Provide any additional context or information that may help us investigate your issue

This would be great to have for people who have multiple networks with multiple VLANS that do not want to have to broadcast their SSIDs to their neighbours to get the Miyoo Mini + connected to WiFi.

therealdickieadams, 2023-04-17:
End user could edit the /appconfigs/wpa_supplicant.conf via the File Browser to create an entry for their hidden SSID.

juggalo16651, 2023-04-24:
I can confirm this is an issue as well. I have multiple vlans in my home and I have a specific vlan so devices are unable to communicate with my main devices, I don't need a cheap Chinese handheld to be on the same network as my personal important data. If I unhide the ssid I am able to connect to it but as soon as I hide it again, it will disconnect and no longer be able to connect. I hide the ssid just so my neighbors don't see multiple networks coming from my home.

Thunderben76, 2023-07-29:
Hello.

Still no solution to connect the MIYOO Mini Plus to a Wifi network with hidden SSID ?

@AceXintense : Have you found a solution to connect your MIYOO Mini Plus to your Wifi network with hidden SSID ?

For the Team Onion : Will it soon be possible to connect your MIYOO Mini Plus to a Wifi network with hidden SSID ?

Cordially.

AceXintense, 2023-07-31:
> Hello.
>
>
>
> Still no solution to connect the MIYOO Mini Plus to a Wifi network with hidden SSID ?
>
>
>
> @AceXintense : Have you found a solution to connect your MIYOO Mini Plus to your Wifi network with [...]

I essentially created a new network on my router which was specifically made for the Miyoo, Hoping that this will be fixed soon!

Thunderben76, 2023-08-01:
Thank you for the return.

At home for Wifi I use a Devolo kit. (the Wifi of my box is deactivated)

Each time I have to deactivate the hidden SSID to be able to use my MIYOO Mini Plus on Wifi, I change before we see my network the name of my network and my password while using the MIYOO Mini Plus over Wi-Fi.

And as soon as I'm done using it on Wifi, I activate the hidden SSID again and then put my "hidden" network name and my password back.

It's very restrictive in the end to have to do this every time.

I don't know why it's taking so long for Team Onion to find a solution to this problem.

tranb3r, 2023-10-30:
Simply edit your `/appconfigs/wpa_supplicant.conf` file and make sure it has `scan_ssid=1` for hidden ssid.
```
network={
        ssid="insert_your_hidden_SSID_here"
        scan_ssid=1
        psk="insert_your_wifi_password_here"
}
```

Thunderben76, 2023-10-31:
Thank you tranb3r.

I didn't have the "scan_ssid=1" line, I added it and now my MIYOO Mini+ can connect to my Wifi with its hidden SSID.

Thanks again for solving this problem.