OnionUI/Onion #782: Crashes and buggy UI

BehroozRezvani, 2023-04-15:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1.4

### Miyoo Firmware Version

202303021817

### Provide a clear and concise description of the issue

Onion OS doesn't start properly
Games do not load properly
Onion OS battery report is broken
RetroArch crashes after 10 seconds
Sleep doesn't work (screen doesn't turn off)
Shutdown from the settings menu doesn't work
and probably many more that I couldn't test yet

### Steps to Reproduce

This is a new device purchase
Latest firmware by Miyoo I believe, and it has a massive number of incompatibility

### Provide any additional context or information that may help us investigate your issue

I will be able to provide more information upon request

Aemiii91, 2023-04-15:
Onion V4.1 is not compatible with Miyoo Mini Plus.

closed 2023-04-15 by Aemiii91

BehroozRezvani, 2023-04-15:
Sorry! should have scrolled a bit more

a2u, 2023-05-11:
@Aemiii91 is miyoo mini plus planned to be supported? I also bought the plus version :( 
Had I known OnionOS was unsupported, I would have taken the regular version.