OnionUI/Onion #786: V4.2.0.2-beta: ScummVM audio possibly using wrong sample rate

vonmillhausen, 2023-04-18:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

V4.2.0.2-beta-4E7C2DB

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

When playing ScummVM games, the output audio is pitched downwards slightly, and runs slightly slowly compared to what the game's audio should actually sound like. It applies to both sample audio (e.g., speech) and music (e.g., Adlib soundtrack). I suspect this is due to a sample-rate configuration issue, although I haven't been able to find the source myself.

### Steps to Reproduce

1. Launch a ScummVM game that plays sound
2. Compare the audio output from Onion OS vs. ScummVM running on a PC

### Provide any additional context or information that may help us investigate your issue

Here's a sample recording from "Lands of Lore: The Throne of Chaos" (DOS CD-ROM version), as recorded from a PC (zipped because this form doesn't allow upload of MP3 files):

[lol_conrad_dos.zip](https://github.com/OnionUI/Onion/files/11262977/lol_conrad_dos.zip)

And here's the same part of the game recorded from a MM+ running 4E7C2DB:

[lol_conrad_onion.zip](https://github.com/OnionUI/Onion/files/11262967/lol_conrad_onion.zip)

The audio has been aligned so the speech starts at more or less the same time if the two files are played simultaneously. If you have a tuned ear, you can hear that the pitch of the speech (and music in the background) is lower from Onion than from the PC; and if your ear can't pick that up, you'll note that the Onion speech finishes way after the PC speech, because Onion's audio is running slow.

vonmillhausen, 2023-06-16:
I just tried the dev build with the stand-alone ScummVM included, and can confirm it does indeed play audio at the correct rate. I can also confirm there's no changes to the Retroarch core.

Given that the core is unchanged, I'll leave it up to other folks here to decide if this issue should be left open until things are resolved there too.

breacher18, 2023-08-10:
I have the same issue with broken sword 2.5 audio is fine in the standalone however now the cursor movement is extremely slow. Increasing the speed in the scummvm setting doesn’t change the speed in game :/

XK9274, 2023-08-10:
> I have the same issue with broken sword 2.5 audio is fine in the standalone however now the cursor movement is extremely slow. Increasing the speed in the scummvm setting doesn’t change the speed in game :/

I've noticed it's slow in the menu but didn't get much further than the intro, is it slow there too? This is probably because of a downscaler we had to code in, otherwise standalone wouldn't run any 800x600 games at all as the source was targeted for low res games only. Will test a new scummvm core soon. Need to speak with schmurtz to see if any changes were done to the RA core first 👍

breacher18, 2023-08-10:
> I've noticed it's slow in the menu but didn't get much further than the intro, is it slow there too? This is probably because of a downscaler we had to code in, otherwise standalone wouldn't run any [...]

Yes it’s the same in game. Fair enough thanks for the fast response :)

dreams897, 2024-01-25:
Broken Sword 2.5 still has the issues with the slow cursor and high pitched audio, any solutions? The Spyfox games also have issues with high pitched voice audio