OnionUI/Onion #787: GBA games not saving

Thewallrudy1, 2023-04-19:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

Installed onion OS 4.2 to Miyoo mini plus tried saving on transferred save file for Pokémon fire red, game did not save and got stuck on save screen( music still playing). Booted up new game for other Pokémon games and got same issues.

### Steps to Reproduce

Start GBA Pokémon game (FRLG or Ruby sapphire) save game through in game menu

### Provide any additional context or information that may help us investigate your issue

At first I though it was the save file I transfered I back up prior to installing Onion OS but when trying to start new games on other pokemon games I hadn’t played before it still wouldn’t save.

retroNUC, 2023-04-25:
I'm seeing some instances of this as well, my case was with Metroid Fusion.

Played the game for a bit from an existing save, stopping by at several in-game save points, then left the console in standby, which turned itself off after 15 minutes. When I turned the console on again and booted the game again, my SRAM save had gone back to a save point before that previous play session (or, more likely, those SRAM writes requests never got wrote out properly).

Miyoo Mini Plus, 4.2.0.2. Had disabled RetroArch options for State AutoSave/AutoLoad, as I prefer to have Hardcore Mode turned on for RetroAchievements and not use save states at all.

null-character, 2023-04-27:
I would try another rom file, as someone on reddit has the same issue and it was with the rom that was included on the SD card with the miyoo mini +. Once they swapped it out for another copy of the rom from another source saving worked.

Olywa, 2023-05-05:
Indeed there are a lot of bad roms on the stock SD card, always better to source your own. 

@Thewallrudy1 (@retroNUC) did the suggestion by @null-character help with this issue were you able to find and test with alternative versions of the roms? If other games (using the same core) are not exhibiting the same issue it would certainly indicate a rom (or save file) specific issue.

retroNUC, 2023-05-07:
> Indeed there are a lot of bad roms on the stock SD card, always better to source your own.
>
> @Thewallrudy1 (@retroNUC) did the suggestion by @null-character help with this issue? Were you able to [...]

The files I was using were No-Intro sets, not the included ones.

Haven't seen it happen since that one instance.

Thewallrudy1, 2023-05-07:
Hi all, 

I finally was able to get a chance at downloading my own roms and was able to fix the issue. So far I've only had this issue with Pokémon games but now know he pe to fix it if I see it in other games.

Thewallrudy1, 2023-05-07:
Thank you all for the support.

Olywa, 2023-05-07:
> I finally was able to get a chance at downloading my own roms and was able to fix the issue. So far I've only had this issue with Pokémon games but now know he pe to fix it if I see it in other games.

Thanks for coming back to confirm. Glad you got it sorted. I'm gonna close this one.

closed 2023-05-07 by Olywa

lfonseca21, 2023-07-19:
Hi I'm having the same issue, but only with the Pokemon games, I can do a save state, but I can't use the ingame save.
I was playing for more than 20 hours, and when I complete the challange of the Elite Four and beat them, the game goes and save by himself, and this make the game crash and go to screen with "save failed checking the backup memory". FYI, the others GBA games and others plataforms works just fine.

I also tried few pokemon versions and few different ROM's, none of those worked.

PLEASE HELP ME ! I bought the console to play Pokemon mostly!

Thewallrudy1, 2023-07-19:
> Hi I'm having the same issue, but only with the Pokemon games, I can do a save state, but I can't use the ingame save.
>
> I was playing for more than 20 hours, and when I complete the challange of the [...]

If you are using the ROMS provided with the system, try to downloading your own and that helped resolved the situation. I'm completely new to all and having search for my own Roma was a bit intimidating at first but a few YouTube searches made it super easy.

lfonseca21, 2023-07-19:
> If you are using the ROMS provided with the system, try to downloading your own and that helped resolved the situation. I'm completely new to all and having search for my own Roma was a bit [...]

Hey @Thewallrudy1 ! Thanks for responding !

Actually I've tried few ROMs of the same game, but unsucessful, do you have any especific website that I can download from?

My GOAL is to have the Pokemon Emerald working. Since I lost my +20h of gameplay. (sad)

Aemiii91, 2023-07-19:
@lfonseca21 no asking for roms

lfonseca21, 2023-07-19:
@Aemiii91 I'm sorry, didn't know!

lfonseca21, 2023-07-19:
Hey @Thewallrudy1 ! Thanks for responding !

Actually I've tried few ROMs of the same game, but unsucessful. 

My GOAL is to have the Pokemon Emerald working. Since I lost my +20h of gameplay. (sad)

ZesPak, 2023-08-12:
Anyone found a solution to this? I also lost a couple of hours of gameplay.
I tried a couple roms but the same issue. I was trying Pokemon Leaf Green and Pokemon Fire Red

The odd thing is, I actually used the save feature in the Fire Red rom before moving to OnionOS, where it worked no issue.
Model MY354 (Miyoo Mini+)
Firmware 202306282128
Running Onion-v4.2.0-RC-snapshot-a5064207

retroNUC, 2023-08-12:
As far as I know, RetroArch doesn't actually save the SRAM to disk until core shutdown, so a sudden interruption without safe shutdown of the application may end up with lost save data. I think there may be cases where the Miyoo OS may not be doing this properly (my money is on idle standby), but it'll be hard to confirm without full RetroArch logs.

This can be mitigated by turning on the "SaveRAM Autosave Interval" option, requesting that the SRAM get saved to disk every X minutes, which both RetroArch and OnionUI do not enable by default. Is it worth adding that the OnionUI RA config patch?

A few weeks back, I spend an evening trying to reproduce this and prove the non-clean shutdown case, but with no luck.

ZesPak, 2023-08-12:
Thanks for that quick reply, tried that on both roms, still this message when trying the in-game save feature.
Does the OnionOS use a different GBC/GBA core than the default OS?
[image]

lfonseca21, 2023-08-12:
In my case I could fix using another ROM, I had to try few to find the right one. In order to not lose my saving I've used my State save to recover and then I save in game. Now, every game that I play I usually test the save first.

ZesPak, 2023-08-12:
> 

Hi, quite interested in how you could recover those states?
I just tried putting them in the folder of the core that worked for me (gsSP), but they seem to be stuck on the screenshot I posted previously.
You switched roms but not cores, if I understand it correctly?

wb14123, 2023-08-20:
@lfonseca21 could you provide more details about how to recover from state save? I copied the state save file from old rom to new rom under `Save/CurrentProfile/states/mGBA`, but the game stuck on the save loading.

lfonseca21, 2023-08-20:
Hi @wb14123, TBH I don't remember exactly the steps I made, but yes, I saved the state for my current game, you can see that there are one for each time that you save, I took the last 2 and uploaded to the state folder in my SD card, when I started the game, I entered the RetroArch and force the game to a specific state (the one that I saved). And boom, it worked!

lfonseca21, 2023-08-20:
And TBH again, I had 0 believe that it would work, but happily it did ! LOL

wb14123, 2023-08-20:
@lfonseca21 Maybe the rom I downloaded isn't compatible with the stock one. Were you playing Fire Red? If so, which Rom version did you download? I found a `v1.1` version online but that doesn't work.

lfonseca21, 2023-08-20:
No, I was playing Emerald for GBA

subcidr, 2025-06-23:
This is still an issue two years later.. 🫤