OnionUI/Onion #791: Gameswitcher closes

Goodingn, 2023-04-22:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.1.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Gameswitcher exits after about 8 seconds, when it should stay active until I exit it or select a game to play.

### Steps to Reproduce

Set the startup to be gameswitcher.

Issue 1: Power on the Miyoo. The gameswitcher shows, but is only functional for about 8 seconds before it exits to the main menu.

Issue 2: Power on the Miyoo. Quickly select a game in the Gameswitcher (before issue 1 occurs). Play game. Press menu button to access Gameswitcher. The Gameswitcher functions for about 8 seconds before resuming the game you were just playing.

If I let Issue 1 play out, i.e. exits to the menu, Issue 2 doesn't occur.

If I set the startup to not be the Gameswitcher, Issue 2 doesn't occur.

Didn't have this issue in v4.0.4

### Provide any additional context or information that may help us investigate your issue

.

emperordahc, 2023-05-03:
I have this exact same issue and my symptoms are exactly as above.

Goodingn, 2023-06-18:
I see this has been marked as 'unable to reproduce'.

Deleting the content_history.lpl fixed this for me too.

I still have a copy of it if someone wants it sending to test?

Goodingn, 2023-06-21:
Shared file [here](https://drive.google.com/file/d/1VWHUm4id50nXk-WETJn26ITQBVjIRyrM/view?usp=drivesdk)

Aemiii91, 2023-07-27:
Can you reproduce this on 4.2-RC ?  
If not, I'll consider this solved.

philippeNiles, 2023-11-28:
I had an issue with the game switcher where I couldn’t see the game screenshots anymore. I noticed that the directory 
`/mnt/SDCARD/Saves/CurrentProfile/romScreens`
Was not actually a directory and just and empty file with no extension. 
I deleted the file while also deleting my `content_history.lpl` and was able to obtain screenshots again. 

The `romScreens` folder is now an actual directory. This might be the v4.2 version of the same issue in v4.1 but not 100% sure.

tGecko, 2023-11-28:
>Was not actually a directory and just an empty file with no extension. I deleted the file while also deleting my `content_history.lpl` and was able to obtain screenshots again.

This sounds very much like a dying SD card

philippeNiles, 2023-11-28:
> > Was not actually a directory and just an empty file with no extension.
> 
> This sounds very much like a dying SD card

I highly doubt it since I bought a new Samsung sd card. Possibly just a corrupt copying onto the sd card when I originally installed the OS. 

But again, deleting the file fixed the issue and I can now see the screenshots without issue.

tGecko, 2024-03-11:
Closing this as GameSwitcher doesn't use `content_history.lpl` anymore.

closed 2024-03-11 by tGecko