OnionUI/Onion #792: Implement cloud saves for the Mini+

schmurtzm, 2023-04-22:


anzz1, 2023-04-22:
A simple way to do this would be to make a scraper program/script, which would connect to the MMP via SSH / SFTP / FTP, and sync those saves back and forth. This would work with the current Onion 4.2 beta already, with no need to change any code. 

A script like that could be used for periodically backing up the saves too, which would be pretty neat, even if not using any other platforms.

retroNUC, 2023-04-26:
Scripts like what anzz1 mentioned are what I've done on past offline consoles, running when SD card is mounted to a PC. It's save sync, but not _cloud_ sync.

If we're doing it properly for cloud sync, I would suggest these elements:

- [rclone](https://github.com/rclone/rclone) for the major work
- OnionUI app for rclone config/setup of the most common services (dropbox, gdrive, onedrive, smb)
    - Optional for now if more advanced users are happy editing config manually
- Running the rclone sync task when...
    - We start up an emulator 
    - We gracefully exit an emulator

jyannick, 2023-04-28:
There is also [synthing](https://syncthing.net/) which is not exactly a cloud sync, more of a local sync. It is supported on many platforms.  Devices have to be up simultaneously to perform the sync, but if one of them is a smartphone that is basically always connected, this is not really a big constraint in practice.

synthead, 2023-05-13:
Head's up!  You probably _do not want credentials saved on the device_ until https://github.com/OnionUI/Onion/issues/818 is resolved, and the fix is deployed to your device :boom:

seanstardev, 2023-12-29:
> Scripts like what anzz1 mentioned are what I've done on past offline consoles, running when SD card is mounted to a PC. It's save sync, but not _cloud_ sync.
>
> If we're doing it properly for cloud [...]

I've hooked the MM+ up with syncthing and it's brought some interesting use cases to light that would affect a cloud save solution.

Ideally - this cloud save implementation would attempt to sync (push and pull saves) as it enters sleep mode and when the console is abruptly turned off, as well. Running sync when the emulator starts is a great idea, but could add a lot of delay to the quick launching of games.

Given the (brilliant) feature of disabling netwok when in game mode, temporarily re-enabling network as we enter 'sleep' and 'shutdown process' could be an involved task .

Further - we'd really benefit from the option of showing sync success / fail notifications when back in a non-game screen.

Finally - an explicit 'sync now' button on the main menu screen would add great piece of mind.

JakiShen, 2024-01-14:
Add Gamelist.xml file generator please

enj, 2025-01-04:
RetroArch has a native cloud sync feature for save files that uses WebDAV, any chance we could get that in OnionOS?  I am not sure which version added it - I think 1.19.x has it enabled by default on Linux.  OnionOS seems to be using a fork of 1.15.x.

Aemiii91, 2025-01-04:
@enj Onion v4.4 updates RetroArch to 1.19.1