OnionUI/Onion #796: miyoogamelist.xml not working on 4.2.0.2

whoozwah, 2023-04-23:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.0.2

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

When adding a properly generated and formatted miyoogamelist.xml to the root of a system rom folder and forcing a rom refresh in the OS the game names do not update to reflect the data in the xml document.

### Steps to Reproduce

Generate a new gamelist.xml via skraper for any given system, run the xmlstartlet script to clean the excess metadata from the xml file and add it to the rom folder for the system, refresh the rom list and verify that the game names have not updated to reflect the changes.

### Provide any additional context or information that may help us investigate your issue

So far here's what I've done:

Used skraper to generate gamelist.xml files for each system I have

opened them with notepad++ (with EOL Conversion to Unix (LF) and encoding to UTF-8) and made desired changes to my game names and saved them to the root of the rom folder for each system.

Ran the script provided in the PDF document found here: https://github.com/OnionUI/Onion/files/9050638/miyoogamelist.20220607.pdf from a normal command prompt window, not powershell and then checked that the extra meta data was removed from each game entry in the xml files.

Ensured that each system's config.json has the applicable extension list for the roms configured

Put the card back in the device, highlighted games in the mainUI and did a "refresh roms" and I got a very short notification that it was clearing all caches.

Check the game list for each system and no changes to the game names are being applied.

To my knowledge that is the full instructions to the letter but it doesn't work. I also did some extra troubleshooting by manually creating my own miyoogamelist.xml from the example code in the PDF guide and saved it with the correct encoding and end of line arguments but that doesn't work.

Finally, I tried removing the card, booting the device to the onboard firmware, reset the device to defaults and then reinserted the card and did a refresh roms again but still no change.

Olywa, 2023-04-26:
I have today verified that the miyoogamlist xml is working correctly in both 4.1 and the 4.2.0.2 beta release.

There is likely still an error in your xmls., can you attach an example of one of your lists? And maybe a screenshot of the system rom folder it relates to (showing the first few roms). 

If you have ARCADE setup with mame roms, you can also try with this example, just extract to `\Roms\ARCADE\` - it's always better to generate your own lists but this should cover all Mame 2003plus roms...
[miyoogamelist.zip](https://github.com/OnionUI/Onion/files/11327640/miyoogamelist.zip)

whoozwah, 2023-04-26:
https://drive.google.com/drive/folders/1aMymlGvoROFIse0xgwikHbu12inqPsx1?usp=share_link

This contains the original gamelist.xml as generated by SkraperUI, the xmlstartlet script I used to delete the excess metadata from it and the resultant miyoogamelist.xml with some edits I made to the game names under the <name> field to what I wanted.

Olywa, 2023-04-26:
OK, I see the issue, you are missing the <image></image> tags for each game. 

I see this is not present in your original file from Skraper ether (it should be populated by default I believe unless you customise something in skraper) but Onion (and the underlying Miyoo MainUI) expect these tags or it will fail. 

It should work even if the image does not exist, as long as the xml is populated. Try this one...
[miyoogamelist.zip](https://github.com/OnionUI/Onion/files/11327750/miyoogamelist.zip)

whoozwah, 2023-04-26:
I see. I did disable all images in skraper because I do not want them. Lemme check this and post back with results.

whoozwah, 2023-04-26:
This works. I have 2 more questions. 

1. Can the neogeo.zip be hidden? I tried putting it in the retroarch/system folder but for some reason the game won't launch unless it is inline with the roms. May be that it's not a non-merged set. I can try to track down a non-merge set if hiding in the list is not possible.
2. Do you have a suggestion for generating new xmls that have the necessary tags in them? images are irrelevant and rescraping with images seems like an unnecessary bandwidth eater

Olywa, 2023-04-26:
Cool, try this...

1) FYI, You mention the /retroarch/system/ folder, Onion does not use this folder, the root BIOS folder replaces it. It's common with neogeo to need the bios file in both the /Bios/ and /Rom/NEOGEO/ folders. Just remove the relevant block of text from your xml and it won't show. 
```
  <game id="152761" source="ScreenScraper.fr">
    <path>./neogeo.zip</path>
    <name>ZZZ(notgame):Neo-geo Bios</name>
    <image>./Imgs/noimage.png</image>
```
2) Ideally some simple scripting but a quicker way would be to just use notepadd++ find and replace to replace all
`</game>`
with
```
  <image>./Imgs/noimage.png</image>
  </game>
```

[image]

closed 2023-04-26 by Olywa