OnionUI/Onion #800: Wifi on/off gone from settings after wifi OTA update

sycookuk, 2023-04-27:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Latest BETA

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

First time I ran the OTA updater it found my wifi straight away and updated my mini. BUT after this update I can no longer turn wifi on and off from settings or have a wifi option in settings to connect to another wifi network?? But the OTA updater works if I'm in the house where I first setup the wifi connection? I want to be able to turn wifi on and off and connect to different network but now cant? 

### Steps to Reproduce

Go to settings if you can see wifi option in there that's where I cant? 

### Provide any additional context or information that may help us investigate your issue

Oddly the OTA update still work and kicks in the wifi connection.

Olywa, 2023-05-05:
It's a longshot but can you try switching themes to see if it restores the wifi setting?

sycookuk, 2023-05-05:
Fixed by starting fresh on a new sd card very odd issue!

closed 2023-05-05 by sycookuk