OnionUI/Onion #808: Miyoo Mini+ Games Crash after opening (consistent)

MrMiklo1, 2023-05-03:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2 beta.dev

### Miyoo Firmware Version

202303362339

### Provide a clear and concise description of the issue

When I open a game and have the title screen play, it crashes back to the game list of whatever emu I am on. When I go to re open it, it'll play a little more (start game, play) then randomly crashes back to the game list. Every time it is opened the spot it crashed at last is returned. Have tried on multiple games, same results. 

### Steps to Reproduce

1. open game
2. play game (wait random time approx. 5 secs)
3. game crashes 
4. repeat steps

### Provide any additional context or information that may help us investigate your issue

Have miyoo mini plus
Run 4.2 beta software
Wifi does not turn on as well

Olywa, 2023-05-05:
This sound like a bad install or corruption isolated to yourself I'm afraid, it's not something we are seeing anyone else report.

I can only suggest a reinstall on a good SD card (if you happen to be use the stock SD card, definitely replace that with one from Samsung, Sandisk or Kingston) and avoid also using the USB card reader that came with the mini, both can cause corruption. 

just to confirm, your firmware is 202303**2**6 right, (not 202303**3**6 as listed above ?

closed 2023-05-07 by Aemiii91

MrMiklo1, 2023-05-07:
Ok, so I figured it out. It was bad roms that came with the original memory card. I bought a PNY 64 gb (overkill on the mem) but since, the games have saved properly.

As for the WIFI, the option finally turned on for some reason and let me connect, which was odd to me considering when I manually tried to turn it on it wouldn't work. 
Yes, 20230326

paulkastel, 2023-05-25:
I am adding comment in case when someone would stumbled on this ticket with the same error. I have the same device, firmware and OnionOS version.

In my case I had already brand SD card (Kingston 64GB - almost fully packed) and games were crashing (across all emulators)... This happened in my case, when battery level was less than 5%. **Charging Miyoo Mini+ to 100%** solved problem for me and crashing stopped.

Looks like battery energy level has impact on read SD card speed.

Aemiii91, 2023-05-25:
@paulkastel in your case it's not a bug, it's the "Save and exit when battery <4%" feature (this can be toggled in Tweaks)