OnionUI/Onion #810: Battery percentage is not correct during recharge

naren-rajendran, 2023-05-08:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

V4.2.0.2-beta-4c7e2db

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

While charging the device, miyoo mini plus, the battery percentage is not displaying correctly or calculated incorrectly. After couple hours of charging the battery percentage was still below 10% (started charging when my device showed 5%).of 

### Steps to Reproduce

- Charge the device (miyoo mini plus in my case) when the battery is below 10%
- Wait for couple of hours(my case) or few minutes
- Remove the cable to check if the battery has more juice than the start of charging (the charge would/should be the same as the start or hasnt increased at all)

### Provide any additional context or information that may help us investigate your issue

n/a

princeprospect408, 2023-05-09:
I've been encountering the same problem all morning. I just got my miyoo mini a few days ago. This is my second charge (at 50%). It has been a few hours now, trying multiple usb-a to usb-c cables and power adapters (including my pc) with no change to the shown charging percentage. There seems to be incosistencies among different themes as well.

Switching from my favorite theme to the original, the percentage showed 100%. Changing back to my favorite theme again showed 50%. Changing back to the original theme once more now showed 85%. 

Changing to another theme that I enjoy showed full battery, then going back to the original theme showed 100% once more. THEN going back to where I originally started, on my favorite sonic theme, it never changed, staying consistent at 50%.

Very strange. Really hope it gets worked on soon, but at least I know other people are having a similar experience and my new unit isn't faulty. Just wanted to share my experience here.

Happy gaming

Joeyjr00, 2023-05-10:
I am currently having the same issue. When plugged in, the border is red and says I have zero percent. The power button also doesnt work when its plugged in. If I unplug the cord, it says im charging. If I turn off it ether shows the charging logo without actually being plugged in. The alternative is that if I turn if off, it will just turn back on.

a2u, 2023-05-11:
Same situation, OnionOS shows 0% battery charge. On the stock it is correct.

Model number: MY283
Firmware version: 202303292051
OnionOS 4.1.4

ena740m, 2023-05-24:
I'm also seeing this issue. However my battery percentage stays at 0%. Same Miyoo firmware version.

mbledkowski, 2023-05-27:
Same issue

duckyb, 2023-05-30:
Alternative steps to reproduce:

1. Start with battery below 100% on the Onion home screen
2. Tap the power button to sleep, leaving the device ON
3. Charge for a while
4. Unplug the device and wake the screen
5. **Observe the battery level is the same as before**
6. Start a game
7. Immedeately exit back to the Onion home screen
8. **Observe the battery level is now updated to the correct value**

alexthestreet, 2023-06-10:
> 

Same issue with a SanDisk 64gb microSD. Any update on a fix?

mindougby, 2023-06-12:
Same problem. Also, the volume buttons do not work, and the application / game closes a few seconds after launch.

Aemiii91, 2023-06-12:
@mindougby sounds like you're using the wrong version of Onion

biuklija, 2023-06-15:
Can confirm, yesterday I found my MMP fully discharged (although I left it 'fully charged' and switched off). 
I was able to switch it on while connected to the charger, launched a game, got a blinking red battery empty overlay a few times and the game would exit to menu after a while. When I disconnected the charger, the OnionOS battery level indicator claimed I was at 80-ish percent, which I clearly wasn't. 
Using Onion-v4.2.0-beta-dev-072c73c.

closed 2023-06-15 by Aemiii91

ericmoore123, 2023-06-28:
So no help on this or what?

Mook252, 2023-07-19:
Same issue here. Battery level permanently at 0% even when it's fully charged. Guess it's back to stock firmware then :/

Aemiii91, 2023-07-19:
@Mook252 and you installed the latest pre-release? (4.2.0-RC)

Mook252, 2023-07-19:
@Aemiii91 Thank you for your reply, I had thought I was on the latest version but I now see that I am not. 
Sorry for not checking properly. I will give the 4.2.0-RC a try.

jemcik, 2024-06-25:
Battery level permanently at 0% even when it's fully charged. 
I am using latest firmware `202306282128` and latest Onion V4.3.1
I also tried reverting to older firmware and to stock OS and nothing helps.

r57zone, 2024-06-25:
It's not the firmware, other firmwares have the same problem. Probably the battery controller is malfunctioning. If you remove the battery cable and plug it back in, the battery level shows correctly, you can try to solder a toggle switch that will disconnect the power to the device physically.