OnionUI/Onion #811: Atari ST (hatari) does not support save states

LukeDearden, 2023-05-10:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion-v4.2.0-beta-dev-4c7e2db

### Miyoo Firmware Version

202303292051

### Provide a clear and concise description of the issue

Atari ST save states do not currently seem to be supported in Onion despite having the same version of the Hatari Core as the RG35XX 

Core version Hatari v1.8 

[image]

I tried setting saves to true and serialised in .info file but it don’t change anything

This was the case on 4.1 on the regular MM too

### Steps to Reproduce

Enabled Atari ST through the Expert menu

Start a game and exit with the menu button as usual to create a save state

starting  the game from the game switcher reloads the content 

### Provide any additional context or information that may help us investigate your issue

Asked to create an issue by Aemiii91 on the MM discord

Olywa, 2023-05-10:
Hmmm, the official [documentation for this core](https://docs.libretro.com/library/hatari/) indicates that states are not supported but the fact you suggest states are working on the RG35XX had me curious and it looks like they can be easily enabled. 

Go to `RetroArch\.retroarch\cores` and replace the contents of the `hatari_libretro.info` file with the contents [from here](https://github.com/libretro/libretro-core-info/blob/master/hatari_libretro.info) then in the same folder, delete the `core_info.cache` file so the changes will be recognised. 

Please confirm if successful and we can make sue this is updated in a future release.

LukeDearden, 2023-05-11:
> 

Yep that did the trick. Saves states now work.

Hoping the Atari ST could get promoted to a regular system(non Expert) some day

Olywa, 2023-05-11:
@schmurtzm Could you update the info file for the next release please?
...and consider moving the emu out of expert.

closed 2023-05-27 by Aemiii91