OnionUI/Onion #813: GameSwitcher functionality preventing RA from setting disc

davidhewett, 2023-05-11:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

V4.2.0.2-beta

### Miyoo Firmware Version

202303292051

### Provide a clear and concise description of the issue

I'm experiencing an issue with multi-disc PSX games, where Retroarch is unable to set the disc to the previously used one in certain conditions dependent on usage of the GameSwitcher (making it seemingly impossible to use the GameSwitcher once on the second disc). The error is "Failed to set last used disc...", which appears on the bottom left corner when (re)booting the game.

### Steps to Reproduce

1) Boot up any multi-disc PSX game (CHD files loaded via m3u)
2) Long press menu button to exit
3) Reopen game
4) Press menu to open game switcher
5) Press A to resume game

"Failed to set last used disc..." should always occur after these steps.

### Provide any additional context or information that may help us investigate your issue

I'm using the Tiny Best Set, and I verified that the m3u is correct. The log file on the root of my SD card had the following errors:

```[ERROR] [Environ]: GET_VARIABLE: pcsx_rearmed_async_cd - Invalid value.```

```
[ERROR] [Disc]: Failed to set initial disk index:
> Expected [1] /mnt/SDCARD/Roms/PS//_hidden/multi-disc/Metal Gear Solid (USA) (Disc 1) (Rev 1).chd
> Detected [1] /mnt/SDCARD/Emu/PSX/../../Roms/PS//_hidden/multi-disc/Metal Gear Solid (USA) (Disc 1) (Rev 1).chd
```

```
[ERROR] [Disc]: Failed to set initial disk index:
> Expected [1] /mnt/SDCARD/Emu/PSX/../../Roms/PS//_hidden/multi-disc/Metal Gear Solid (USA) (Disc 1) (Rev 1).chd
> Detected [1] /mnt/SDCARD/Roms/PS//_hidden/multi-disc/Metal Gear Solid (USA) (Disc 1) (Rev 1).chd
```

Levistras, 2023-08-10:
I see this message often when gameswitch into ff7/psx, but I'm still on disc 1 so it hasn't impacted gameplay.

QuackWalks, 2025-04-23:
This is an issue with tiny best set and the .m3u files it provides. The hidden folder should be named .hidden and the .m3u files should point to .hidden/