OnionUI/Onion #814: Theme creation, bg-title repeating pattern

nithou, 2023-05-12:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

When creating a theme and adding a logo inside bg-title.png, if the device is plugged in and charging, the bg-title seem to adopt a repeating on x pattern which duplicates it behind the wifi and battery icons if you don't use non transparent icons

### Steps to Reproduce

Add a logo to the left of bg-title.png and make wifi and battery icon with transparent background. Plug the device for charging, your logo will partially appear behind the wifi and battery icons

### Provide any additional context or information that may help us investigate your issue

At the moment we can only use miyoo-topbar to put a logo but then it doesn't appear on list, settings and things. We should be able to use bg-title for this but the repeating pattern seem weird (issue experienced with my theme if you put a logo on the bg-title file: https://github.com/nithou/minimal-onionOS)

Aemiii91, 2023-05-12:
Please provide some example screenshots of the issue. I'm pretty sure it's a MainUI thing (the closed source frontend by Miyoo - we can't really change much there)

Aemiii91, 2023-05-12:
Thanks! Actually seems like something I might be able to fix 👌