OnionUI/Onion #818: Miyoo Mini Plus & V4.2.0.2-beta: telnetd running by default with no authentication, and it hosts a root shell

synthead, 2023-05-13:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

V4.2.0.2-beta

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

By default with no changes, Onion is listening for telnet connections:

```
$ nmap -Pn 10.1.0.169
Starting Nmap 7.93 ( https://nmap.org ) at 2023-05-12 22:43 PDT
Nmap scan report for 10.1.0.169
Host is up (0.012s latency).
Not shown: 999 closed tcp ports (conn-refused)
PORT   STATE SERVICE
23/tcp open  telnet

Nmap done: 1 IP address (1 host up) scanned in 11.14 seconds
```

What's worse is that it does not use any authentication:

```
$ telnet 10.1.0.169 23
Trying 10.1.0.169...
Connected to 10.1.0.169.
Escape character is '^]'.

/ #
```

And what's even worse than that is that it is a root shell:

```
/ # whoami
root
```

This is the worst-case scenario for security!  With this in place:

- Anyone who has access to the device can log in
- Anyone can read any files on the system
- Anyone can write any files on the system
- Anyone can sniff traffic and monitor anything on the system
- Anyone can use the device as a bastion host for networks
- Telnet is often used by bots that attempt to log in to vulnerable systems like these

While this may seem somewhat mild because it's a little handheld, this means that:

- If you have a remote host that backs up save states, the credentials can easily be stolen
- Someone can log in and erase any data as they please
- If the Miyoo is connected to a VPN, assume that the VPN certificates have been compromised and understand that an unknown third-party could have been on the network the VPN provides
- If any passwords have been stored, they could be compromised
- Unknown third-party software could be installed on the device

Even if there was authentication on the telnet connection, telnet is not encrypted, so the entire telnet session can also be sniffed and exploited.

I greatly advice telnetd to _not be running_, and it should _absolutely not serve root access_.

### Steps to Reproduce

1. Connect the Miyoo Mini Plus to a network
2. Log in via telnet: `telnet 1.2.3.4 23`
3. Observe that full root access is immediately given to the user with no credentials over an unencrypted connection

### Provide any additional context or information that may help us investigate your issue

n/a

synthead, 2023-05-13:
Related Reddit thread: https://old.reddit.com/r/MiyooMini/comments/11yy5ue/dont_connect_to_public_wifi_with_your_new_mm/

XK9274, 2023-05-13:
This is being addressed in a coming release where a toggle will be added that will persist across boots and will kill telnetd when it comes up. It's firmware related but will be resolved. See the PRs

synthead, 2023-05-13:
> See the PRs

Could you link which PRs you are referring to?  I think you're referring to this one, but it was closed without being merged:

- https://github.com/OnionUI/Onion/pull/799

...and there's this one, but it seems to add a menu option that doesn't do anything:

- https://github.com/OnionUI/Onion/pull/807

XK9274, 2023-05-13:
The latter, i was on mobile so couldn't link. Have you copied files to the SD? If so which files?

  *   #799<https://github.com/OnionUI/Onion/pull/799>

  *   #807<https://github.com/OnionUI/Onion/pull/807>

anzz1, 2023-05-13:
I put up a light version of dropbear (191KB , < 2MB RSS) with a master password option here: https://github.com/anzz1/dropbear-embedded/tree/miyoomini , even using the fastest Blowfish cipher it's still slower than Telnet though, but it's secure. It also allows using the 'none' cipher and mac so stuff like rsync could be used fast, without encryption, in a safe environment.

run with `./dropbear -R -Y mypassword`

synthead, 2023-05-15:
> The latter, i was on mobile so couldn't link. Have you copied files to the SD? If so which files?

What do you mean by "have I copied files?"  Are you asking what ROMs are on my personal Miyoo Mini+?

> I put up a light version of dropbear (191KB with -O3, < 2MB RSS, < 100KB with -Os but -O3 is preferred for speed) with a master password option here: https://github.com/anzz1/dropbear-embedded/tree/miyoomini , even using the fastest Blowfish cipher it's still slower than Telnet though, but it's secure. It also allows using the 'none' cipher and mac so stuff like rsync could be used fast, without encryption, in a safe environment.
> 
> run with ./dropbear -R -Y mypassword

I would greatly discourage anyone to disable SSH ciphers by default.  I would go as far as to not support this path, or even mention it in any documentation.  It's just a bad idea in general.

The performance you mentioned also implies that there is a bottleneck that produces performance poor enough to warrant sacrificing security, and that the bottleneck is CPU-bound.  The Miyoo Mini+ uses a 1.2 GHz ARM SoC CPU, which is certainly enough kick to run ciphers we've been using for decades.  The bottleneck, if concerned, would originate from the 2.4 GHz 802.11 network, or the generally-poor I/O performance found on SD cards (as compared to SSDs, for example).

Worse, though, it promotes poor security practices.  Anyone that can sniff network traffic can capture an unencrypted SSH connection, i.e. keystrokes and any files sent or received.  In addition, if you set a password via `-Y password`, this means that the password will be visible by any user that runs `ps -ef`, that the password is stored in plaintext by some means, and that you're using password auth in the first place (when keys are much better).

In my opinion (and feel free to disagree here), we should set users up for success first.  "Don't connect it to anything but your home network" is not a great security policy, in my opinion.  We cannot guarantee that people simply won't do this, and that users will never be vulnerable.  At heart, it's a mobile device with wireless networking, and it's not a good idea to leave it so vulnerable to attacks and theft of data.

This is important in my book because, naturally, people will want to exercise the wireless nature of the Mini+ to do creative things like back up save states (see https://github.com/OnionUI/Onion/issues/792).  If they wanted to back up their stuff to Google Drive, for example, then they would put Google credentials on their Mini+.  Everyday users don't always have security as their #1 priority, so it's very reasonable to expect at least some users to allow these credentials to auth with Google apps beyond Drive.  If they happen to give access to Gmail, now an attacker can use their Gmail account to perform password recoveries and access their bank, for example.

This all seems silly and blown out of proportion simply because the Mini+ is a little Game Boy-looking thing that appears innocent.  However, it is a full embedded Linux device with a squashfs filesystem, just like most "home routers" that runs firewalls as the first step of security for someone's internet connection, and is the same type of system that is used for most of the internet.  Therefore, bots that run scripts to attack vulnerable devices are absolutely targeting devices like these, and it is not good practice to remain vulnerable by choice.  Security policies like "keep your wifi off" or "don't leave your house" are policies that encourage users to get hacked.

anzz1, 2023-05-15:
> > The latter, i was on mobile so couldn't link. Have you copied files to the SD? If so which files?
> 
> What do you mean by "have I copied files?" Are you asking what ROMs are on my personal Miyoo Mini+?
> 
> > I put up a light version of dropbear (191KB with -O3, < 2MB RSS, < 100KB with -Os but -O3 is preferred for speed) with a master password option here: [anzz1/dropbear-embedded@`miyoomini`](https://github.com/anzz1/dropbear-embedded/tree/miyoomini?rgh-link-date=2023-05-15T18%3A46%3A31Z) , even using the fastest Blowfish cipher it's still slower than Telnet though, but it's secure. It also allows using the 'none' cipher and mac so stuff like rsync could be used fast, without encryption, in a safe environment.
> > run with ./dropbear -R -Y mypassword
> 
> I would greatly discourage anyone to disable SSH ciphers by default. I would go as far as to not support this path, or even mention it in any documentation. It's just a bad idea in general.
> 
> The performance you mentioned also implies that there is a bottleneck that produces performance poor enough to warrant sacrificing security, and that the bottleneck is CPU-bound. The Miyoo Mini+ uses a 1.2 GHz ARM SoC CPU, which is certainly enough kick to run ciphers we've been using for decades. The bottleneck, if concerned, would originate from the 2.4 GHz 802.11 network, or the generally-poor I/O performance found on SD cards (as compared to SSDs, for example).
> 
> Worse, though, it promotes poor security practices. Anyone that can sniff network traffic can capture an unencrypted SSH session, i.e. keystrokes and any files sent or received. In addition, if you set a password via `-Y password`, this means that the password will be visible by any user that runs `ps -ef`, that the password is stored in plaintext by some means, and that you're using password auth in the first place (when keys are much better).
> 
> In my opinion (and feel free to disagree here), we should set users up for success first. "Don't connect it to anything but your home network" is not a great security policy, in my opinion. We cannot guarantee that people simply won't do this, and that users will never be vulnerable. At heart, it's a mobile device with wireless networking, and it's not a good idea to leave it so vulnerable to attacks and theft of data.
> 
> This is important in my book because, naturally, people will want to exercise the wireless nature of the Mini+ to do creative things like back up save states (see #792). If they wanted to back up their stuff to Google Drive, for example, then they would put Google credentials on their Mini+. Everyday users don't always have security as their top priority, so it's very reasonable to expect at least some users to allow these credentials to auth with Google apps beyond Drive. If they happen to give access to Gmail, now an attacker can use their Gmail account to perform password recoveries and access their bank, for example.
> 
> This all seems silly and blown out of proportion simply because the Mini+ is a little Game Boy-looking thing that appears innocent. However, it is a full embedded Linux device with a squashfs filesystem, just like most "home routers" that runs firewalls as the first step of security for someone's internet connection, and is the same type of system that is used for most of the internet. Therefore, bots that run scripts to attack vulnerable devices are absolutely targeting devices like these, and it is not good practice to remain vulnerable by choice. Security policies like "keep your wifi off" or "don't leave your house" are policies that encourage users to get hacked.

Blah, blah, blah...Lot of words, with only a little more than zero understanding what you are talking about. You make wild assumptions and guesses which you *think* are correct because they *sound that they could be*, while only having the surface-level understanding of the underlying concepts which allows you to make these guesses but not understand why they are wrong, which is the worst position to be in, understand just enough to lead people astray but not enough to actually know anything.

First of all, I'm not talking out of my ass here like you are. I don't "imply" anything, I have tested it. Doing crypto on SW is hard on the CPU. That is not an assumption, unlike the ones you are making, that is a fact. You don't have to take my word for it though, you can try it yourself. Get a SSH server running on the device, and choose some recent uber-secure algos, like `chacha20-poly1305` or `aes256-gcm` for the transport cipher, `hmac-sha2-256-etm` for the integrity mac, and `ecdsa-sha2-nistp521` or `curve25519-sha256` for the key-exchange. Then start rsyncing some files or whatever to get some packets going and tell me the CPU usage of the SSH server process during it. Yes, it does bottleneck, not being able to switch packets fast enough to saturate even the 802.11n wifi connection and definitely not the read speed of the SD card controller.

The product sheet for the [SSD202D SoC](https://github.com/anzz1/miyoomini-resources/blob/master/doc/SSD202D_pb_S_v01.pdf) mentions support for HW crypto which would make the encryption essentially free in terms of compute time. However the kernel shipped with miyoo mini doesn't have the crypto module compiled, probably for not needing / saving space reasons, so that was an dead end. The RSA driver is there, but its useful only for signing keypairs which isnt the crux of the cpu usage, as the tra [... cut ...] w in `ps` nor `/proc/*` or anywhere else. It'll show -Y, but not the password. You would know this if you had actually tried any of this out.

Basically, it seems you've watched one or two youtube videos then asked ChatGPT to write you a post which makes you sound smart. It does not, not to anyone who actually knows their shit. You might sound to those who don't understand the subject matter, who you are leading astray.

Next time, please weigh your own level of expertise before giving critique.

synthead, 2023-05-17:
Well, this is already off-topic, and honestly...

> Blah, blah, blah...Lot of words, with only a little more than zero understanding what you are talking about.

> I'm not talking out of my ass here like you are.

> I am pretty confident you couldn't even carry out such an attack you're so sure of, else you wouldn't have made such a stupid post.

> But like I said, I'm pretty you cannot since your post shows your level of understanding clearly is more of a "I watched a youtube video" than "I've actually done this".

> It is also very clear that you don't understand the environment at all, nor that you have actually tried any of the stuff you're talking about since you would've already proved your own BS wrong by doing that.

> Basically, it seems you've watched one or two youtube videos then asked ChatGPT to write you a post which makes you sound smart. It does not, not to anyone who actually knows their shit. You might sound to those who don't understand the subject matter, who you are leading astray.

> Next time, please weigh your own level of expertise before giving critique.

...I have no interest in responding to you, @anzz1.  So this comment is not a reply to you.  This is a reply to clear things up for users that might mistake the "yelling louder" approach as confidence.

A good engineer knows what they're talking about.  A great engineer works well with others, and communicates well, too.  An excellent engineer goes beyond that and uses empathy to raise others up, share knowledge, and is a kind and supportive person who takes other people under their wing as they advance their career.

To those that observed the above comment, this is not normal or healthy.  It is toxic behavior, with emotionally-driven attacks.  Instead of asking questions to clear things up, imaginary situations were assumed, along with assuming that I'm "stupid."  If my points can be obscured to mean something else that's besides the point, and I can be put down enough, then it's easy to be wrong, right?

I'm not going to try to "talk myself back up," because this contributes to the toxic behavior.  Instead, I'm going to point out things in brevity so people have the right idea.  I don't care about encryption algorithms or how fast I can transfer 2GB of data with SoC CPUs.  I know them well also, but it doesn't matter for this context.  I am concerned about things on a much higher level from a user's point of view.

Here are my basic points:

- **Security guidelines are just that: guidelines, and they are not always followed.**  If the current guideline is to not connect your device to networks that are not your own, then some people will inevitably do it.
- **Code examples will also be followed blindly by at least some users.**  Most folks don't understand the intricacies of encryption, so if there's a code example hanging out there that "works," then some people will use it.  If you tell them that they need "ecdsa-sha2-nistp521" something or rather, they're going to not know what that means and ignore it.
- **Security is not well-understood by a good handful of users.**  Most people don't understand what the difference between "good" and "bad" security is, and most don't care to be educated.  In terms of this project, most people just want to play games.  If they have bad security practices, but they're playing games, then they don't see a need to change anything, nor do they know how to.
- **Security should generally be the responsibility of the project.**  Configure your stuff to use encryption by default.  Avoid default keys and passwords.  Set up users for success first, and if they make their security worse, that should be something they choose.  Compare this to consumer "home routers" where WPA2 is on, and the network password is unique to each access point.  They used to come with no wireless encryption as the default, and it was a small privacy nightmare.

As the Miyoo currently stands, as long as you trust the security your own home network and trust the software your workstations and servers are running, then no third parties _should_ have access to your Mini+ from within your home network.  I also happen to have IoT devices that run telnet on my own network, and I believe that they're safe.  This isn't really a concern.

The biggest problem is that it's a _mobile_ IoT device.  It could get lost or stolen, it'll likely be connecting to various networks as you travel with it, and if you're doing anything interesting on it, you'll probably have credentials on the device, too.  Most users will not use LUKS, GPG, etc. for their own stuff, because that requires effort, knowledge, practice, and ultimately, inconvenience.  Most users will just plop passwords and keys on the Micro SD card in plain text.  And most folks won't know or care how telnet works, or what it even is.

So again, the answer could be "never leave your house with it," and that will mostly work in principle.  In practice, people will ignore this warning and take it with them to various places, and that is where the problem lies.  On the original Mini, there wasn't any need to add credentials, because it didn't have the hardware to connect to anything.  With the Mini+, it is possible to go from "Game Boy emulator machine" to having unknown third parties attack and access your stuff, and the telnet thing makes it really, really easy to make that leap.

While there are certainly other devices like the Mini+ that people travel with, the original point of this issue is to point out that the Mini+ has a uniquely terrible security vulnerability around telnet that should be addressed.  Most users won't know what's going on, and it's a really easy target for attacks.  It's basically the perfect trap since it's such an innocent-looking mobile device that has such a wide audience of users, you know?

junkaddr, 2023-05-21:
> Next time, please weigh your own level of expertise before giving critique.

Someone helpfully reported that telnet is enabled by default then gives reasonable advice to not disable ciphers. You immediately write a 20 page polemic attacking this person...because...I'm not sure? Take a step back and try to be nice to people.

Anyway, to get this back on topic, yes, OnionOS should disable this to have a more secure default. If users want to enable it on their own network, sure, that's fine. If it didn't get disabled, I don't think much would happen aside from friends pranking each other on a network. I seriously doubt "adversaries" are going to setup automated exploit servers on public networks to target an obscure budget Chinese handheld running an even more obscure OS.

closed 2023-05-29 by Aemiii91