OnionUI/Onion #833: Content Directory Override Save Error

snjcky, 2023-05-19:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.2.0-beta-dev-4c7e2db

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

When I try to save a content directory override file, it saves it as filename: ..cfg. This becomes an issue when trying to save/load multiple override files for content that use the same core (ie GB/GBC w/Gambatte or GG/MD w/PicoDrive).  

Anyone know if there's a fix to this?

### Steps to Reproduce

Games > Any Console > Rom Loaded > RA Quick Menu after video+overlay settings are changed > Overrides > 'Save Content Directory Overrides = a ".".cfg file saved to the 'Active Override File' section

### Provide any additional context or information that may help us investigate your issue

I have tried manually renaming the files to the content directory name but that causes RA to ignore the file. It seems to specifically be looking for a ".".cfg file.

I've tried this on other versions of RA (windows) and the problem doesn't occur.

closed 2023-05-29 by Aemiii91

dre10g, 2023-07-19:
I have this same issue on the latest RC release as well.

Aemiii91, 2023-07-19:
The issue is caused by the paths in your `miyoogamelist.xml` starting with "`./`" - a workaround is to remove these prefixes... but I've fixed it here: https://github.com/OnionUI/Onion/commit/b150ae480899d182b5776346b5bd4950ce2861f7 (for next release)

dre10g, 2023-07-19:
Awesome.. thanks for the quick reply!

So for a temporary workaround.. do I just delete the "./" for all rom paths in each miyoogamelist.xml file?