OnionUI/Onion #840: lemiyoo.cn/upgrade link is dead

mbledkowski, 2023-05-27:
### Checklist

- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and something is incorrect or missing

### Wiki Section

https://github.com/OnionUI/Onion/wiki/Installation

### Description

lemiyoo.cn/upgrade link in firmware upgrade section is dead.

### Correction

I have no idea where to find the latest firmware, but there is some available at lomiyoo.com

Aemiii91, 2023-05-27:
We're not linking to lemiyoo.cn ??

Aemiii91, 2023-05-27:
The installation guide has no dead links.

mbledkowski, 2023-05-27:
[image]
[image]
@Aemiii91 We're linking !!

Aemiii91, 2023-05-27:
Then you should have stated that the "firmware upgrade guide" we're linking to is out-dated.

Aemiii91, 2023-05-27:
OK, fixed (https://github.com/OnionUI/Onion/wiki/Installation/_compare/791cfa35427136e50526886be3415a9edf72eb67)

closed 2023-05-27 by Aemiii91