OnionUI/Onion #845: Freezing/Locking up when pausing using game switcher, toggling Wi-Fi and resuming game

HACKRCHILD, 2023-05-29:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion V4.2.0 beta 3 

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

When using the game switcher to pause the state of the game to toggle the Wi-Fi, upon trying to resume the game the console freezes with a black screen. I'm using retro achievements, so possibly this is causing an issue when it's trying to load. Hard shutdown is required. It only seems to do it if you are turning the Wi-Fi into the on state. Otherwise, turning off the Wi-Fi allows you to resume.

### Steps to Reproduce

Make sure you are signed in to retroacheivements and the wifi is turned off.  Launch a game (i was testing using mGBA and gambette), pause the game using the menu button and press b to go back to the main menu. Head to settings and turn on the wifi. Press the menu button again. This should just freeze/leave you on a black screen.

### Provide any additional context or information that may help us investigate your issue

Note that the reverse does not freeze the console (pausing, turning OFF the wifi, resuming).

Aemiii91, 2023-05-29:
Which emulator and game?

Aemiii91, 2023-05-29:
This seems to be an issue with the game state (with cheevos enabled) expecting an internet connection. As it stands, this does not seem like a bug in Onion. Can you reproduce this issue on the PC version of RetroArch?

HACKRCHILD, 2023-05-29:
Hey just tried on the PC retroarch and could not reproduce. To add to this (and to why i think its an onion bug) the freeze does not happen if you resume the game without using the game switcher. Here's two other ways i tried:

- With wifi off and retroarch signed in.... press menu to pause game using game switcher -> turn on wifi -> select the game from games in the main menu = NO FREEZE 
-  With wifi off and retroarch signed in.... press menu to pause game using game switcher -> turn on wifi -> launch retroarch and load rom = NO FREEZE

HACKRCHILD, 2023-05-29:
To add, i have had it freeze when trying to pause the game using game switcher when toggling wifi also. 

Using the retroarch menu to close the game prevents this freezing. 

It seems the freeze is coming from the gameswitcher menu...

Aemiii91, 2023-05-29:
Okay, interesting. GameSwitcher should launch the game the exact same way as the menu.

HACKRCHILD, 2023-05-29:
Yeah i figured as much. I tested it on both my MM+ and did a fresh onion install for good measure. I can replicate this issue on both consistently and can avoid it consistently with the above methods. 

Also i have just tested and with retroacheivements disabled, game switcher still crashed after a wifi toggle when resuming. I think something must be going on with the wifi and game switcher rather than retroacheivments.

HACKRCHILD, 2023-05-29:
It seems that even if you turn the the wifi on and then straight off it breaks the gameswitcher. I will note that it will do a normal shutdown holding the button when it does black screen, you can feel it vibrate.

Could you point me in a general direction in the source code? happy to have a dig. im a c/cpp dev.

Aemiii91, 2023-05-29:
I don't think there's anything particular in the source causing this, it's more likely a memory issue caused by `wpa_supplicant` and `MainUI`. When we launch GameSwitcher from the menu (by pressing `MENU`) we force kill `MainUI`, causing some memory to not get freed properly.

Some notable waypoints in the repo:
- [**Main runtime loop - launch game**](https://github.com/OnionUI/Onion/blob/a62e6b6efa81f0ab81ba3aeb010679c3fbd5ee65/static/build/.tmp_update/runtime.sh#L301)
- [**Kill MainUI method used by `keymon`**](https://github.com/OnionUI/Onion/blob/a62e6b6efa81f0ab81ba3aeb010679c3fbd5ee65/src/common/system/system_utils.h#LL20C9-L20C9)
- [**GameSwitcher RACommand**](https://github.com/OnionUI/Onion/blob/a62e6b6efa81f0ab81ba3aeb010679c3fbd5ee65/src/gameSwitcher/gameSwitcher.c#L378)

HACKRCHILD, 2023-06-20:
To update, issue still present in latest beta. Haven't had a chance to look at the source yet. Will have a dig this weekend.